AllskyTeam/allsky #87: Power requrements and config.txt (max_usb_current=1)

pterodyne, 2019-04-26:
is anyone using max_usb_current=1 in their config.txt?  I just read about it.  Supposedly on a Pi 2 or greater it enables 1.2Ah instead of 600mAh on the usb ports.  Since we already have to have a sufficient supply to power the ZWO, I don't see why this should not be enabled for the Allsky.

pterodyne, 2019-04-26:
Just answered my own question.  Looks like it's set at 1.2Ah by default for the pi 3 and newer.  But this is something to look into for Pi2 and older and the zero. (Explains why my ASI120mini didn't even show up with the ZeroW.)

thomasjacquin, 2019-04-26:
Ah! I didn't know about that option. Thanks for mentioning it. That will probably help a few users with power issues.

EricClaeys, 2021-09-05:
This issue is from a much older release of the AllSky software and the issue appears to have been resolved.  If you think it might still be relevant, please test it with the newest version and submit a new issue if needed.  Thanks

closed 2021-09-05 by EricClaeys