AllskyTeam/allsky #89: Image storage folder

bbillp, 2019-05-15:
The new Raspbian load and ALLSKY installation went smooth but where are the images stored ?   I have /pi/allsky/image.jpg but no storage folder .  I’ve checked all folders including /.xxx .

For now I’ll use Filezilla to upload but will eventually enable FTP within allsky.   Thanks .... 

Pi 3B, Raspbian as of today, ALLSKY as of today and ZWO 178MC.

bbillp, 2019-05-15:
Ignore, the image directory finally appeared after some delay.

closed 2019-05-15 by bbillp

thomasjacquin, 2019-05-16:
Storage only happens for night images. If it's day time, it overwrites the same image (for preview and upload) but doesn't store it.

bbillp, 2019-05-16:
24 hours later ALLSKY works almost perfect so far but regarding the MP4, what software reads the file in Windows 7 or Raspbian, I tried Windows 7 media player with new codecs, QuickTime (win7) and VLC (win7).   Is it possible to save video in another format ?   

Can you give some hints for setting up a FTP host in Win 7 ?  I use Filezilla to upload from my other Pi cameras.  

PS: all night fotos of fog rolling in are also pretty interesting ..... Bill ...

thomasjacquin, 2019-05-16:
The codec is x264. You can read the videos with VLC media player on windows.

I haven't tried but I believe you can replace the hostname (in `ftp-settings.sh`) by the IP of your Win 7 computer so that the images are uploaded to your computer instead of a web host.

thomasjacquin, 2019-05-16:
See this post for how to store files locally: https://github.com/thomasjacquin/allsky/issues/56#issuecomment-455803438