AllskyTeam/allsky #93: Problems with web interface

ossvale, 2019-06-10:
Hi Thomas,
I have some problems in the web interface of your AllSky project (wonderful by the way).
I intsalled the web gui 3 times now, but as I try to connect to the web page http://allsky.local (or to the 192.168.1.11 IP (the one of my raspberry))  it says 
This site can’t be reached allsky.local refused to connect. 
Tried to uninstall and reinstall the gui many times but without luck
Any ideas?
thanks 
Valerio

thomasjacquin, 2019-06-10:
Hi Valerio,

Can you plug a monitor to the Raspberry pi and open a browser and go to http://localhost ?
That will help figuring out the issue. It should come up with the GUI.
If it doesn't then we'll look into why it's not installed properly (check error messages during install).

ossvale, 2019-06-10:
localhost doesn't work.
I installed the GUI again and it seems to have problems on php7.0
I have raspberry pi3 and I updated all the packages before installation 

Here is the output of the GUI install

pi@allsky:~/allsky $ sudo gui/install.sh

****************************************************************
*** Welcome to the Allsky Administration Portal installation ***
****************************************************************

* Installation of the webserver

Hit http://linux.teamviewer.com stable InRelease
Get:1 http://mirrordirector.raspbian.org jessie InRelease [15.0 kB]
Hit http://archive.raspberrypi.org jessie InRelease
Hit http://linux.teamviewer.com preview InRelease
Hit http://linux.teamviewer.com stable/main armhf Packages
Hit http://mirrordirector.raspbian.org jessie/main armhf Packages                              
Hit http://archive.raspberrypi.org jessie/main armhf Packages                                  
Hit http://archive.raspberrypi.org jessie/ui armhf Packages                                    
Hit http://mirrordirector.raspbian.org jessie/contrib armhf Packages                 
Hit http://mirrordirector.raspbian.org jessie/non-free armhf Packages                
Hit http://mirrordirector.raspbian.org jessie/rpi armhf Packages
Hit http://linux.teamviewer.com preview/main armhf Packages        
Ign http://archive.raspberrypi.org jessie/main Translation-en_US                               
Ign http://archive.raspberrypi.org jessie/main Translation-en                                  
Ign http://archive.raspberrypi.org jessie/ui Translation-en_US                                 
Ign http://archive.raspberrypi.org jessie/ui Translation-en                                    
Ign http://linux.teamviewer.com stable/main Translation-en_US                                  
Ign http://linux.teamviewer.com stable/main Translation-en                                     
Ign http://linux.teamviewer.com preview/main Translation-en_US                                 
Ign http://linux.teamviewer.com preview/main Translation-en                                    
Ign http://mirrordirector.raspbian.org jessie/contrib Translation-en_US            
Ign http://mirrordirector.raspbian.org jessie/contrib Translation-en
Ign http://mirrordirector.raspbian.org jessie/main Translation-en_US
Ign http://mirrordirector.raspbian.org jessie/main Translation-en
Ign http://mirrordirector.raspbian.org jessie/non-free Translation-en_US
Ign http://mirrordirector.raspbian.org jessie/non-free Translation-en
Ign http://mirrordirector.raspbian.org jessie/rpi Translation-en_US
Ign http://mirrordirector.raspbian.org jessie/rpi Translation-en
Fetched 15.0 kB in 7s (1,878 B/s)                                                              
Reading package lists... Done
Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package php7.0-cgi
E: Couldn't find any package by regex 'php7.0-cgi'
gui/install.sh: line 17: lighty-enable-mod: command not found
Failed to restart lighttpd.service: Unit lighttpd.service failed to load: No such file or directory.

* Configuring lighttpd

* Changing hostname to allsky

* Setting avahi-daemon configuration

* Adding the right permissions to the web server

* Retrieving github files to build admin portal
Cloning into '/var/www/html'...
remote: Enumerating objects: 27, done.
remote: Counting objects: 100% (27/27), done.
remote: Compressing objects: 100% (21/21), done.
remote: Total 1767 (delta 10), reused 19 (delta 6), pack-reused 1740
Receiving objects: 100% (1767/1767), 2.72 MiB | 993.00 KiB/s, done.
Resolving deltas: 100% (604/604), done.
Checking connectivity... done.
mkdir: cannot create directory ‘/etc/raspap’: File exists

* Modify config.sh

The Allsky Portal is now installed
You can now reboot the Raspberry Pi and connect to it from your laptop, computer, phone, tablet at this address: http://allsky.local

Do you want to reboot now? [y/n] n

thomasjacquin, 2019-06-10:
Ok, that's an issue that happens on older raspbian systems but it's easy to fix. You just need to open `GUI/install.sh` and replace php7.0-cgi by php5.0-cgi
Then you can run the GUI install script again.

ossvale, 2019-06-10:
mmmm same error,... but regarding package php5.0-cgi

Reading package lists... Done                                                                  
Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package php5.0-cgi
E: Couldn't find any package by regex 'php5.0-cgi'
gui/install.sh: line 17: lighty-enable-mod: command not found
Failed to restart lighttpd.service: Unit lighttpd.service failed to load: No such file or directory.

thomasjacquin, 2019-06-10:
Can you try `php5-cgi` ?

ossvale, 2019-06-10:
yes with php5 it worked!
from inside the raspberry I can reach it with localhost, but can I reach it from another computer in the same network?
I tried from my laptop with 192.168.1.11 or allsky.local but it doesn't work

thomasjacquin, 2019-06-10:
Did the IP of your raspberry Pi change? You can run `ifconfig` in a terminal on the Pi to know your IP. Then try to access from another computer using that IP in a browser.

ossvale, 2019-06-10:
the problem was that it tried to connect via https but it's only http!

One last thing... I see that during installation it says version 0.5 (I used the command
git clone https://github.com/thomasjacquin/allsky.git
but I read somewhere there is a version 0.6

Can I see which version is installed now?

thomasjacquin, 2019-06-10:
You've got the latest code. I just forgot to update the version number in `capture.cpp` line 203. So you should be on v0.6 even though it says 0.5 in the terminal.

ossvale, 2019-06-10:
thanks for everything and again great work!
I will let you know when I finish my allsky project

bnred777, 2019-08-16:
I don't know a lot about RasberryPi or Linux but I was having problems too and it was because I was on strech. I upgraded to buster and it helped. I couldn't get anything to work on strech.

EricClaeys, 2021-09-14:
This issue is from an older release of the AllSky software and appears to be fixed.  If you think it might still be relevant, please test it with the newest version and submit a new issue if needed.  Thanks

closed 2021-09-14 by EricClaeys