AllskyTeam/allsky #94: how to set shorter delay between images when using auto exposure?

rave3c0, 2019-06-24:
Hello Thomas, 

Is there a way to set a shorter delay between images when using auto exposure?  

It's taken a while but I had noticed that the images during the night have up to a 30s time gap between them and I have realized that its the auto exposure settings.  

I see in the documentation info for Autoexposure = "Set to 0 to disable auto-exposure at night. Auto-exposure delivers properly exposed images throughout the night even if the overall brightness of the sky changes (cloud cover, moon, aurora, etc). **When set to 1, maxexposure value will be used as the delay between timelapse frames.** "  

My max exposure is usually set to 30s which explains the bigger gaps when its super dark.  I really like the auto exp as the images and timelapse come out great.  

Thanks for the help!  keep up the great work.
- Bill C.

thomasjacquin, 2019-06-27:
Hi Bill, this is by design because otherwise, the frame rate of the video wouldn't be constant. The keograms would also be stretched in a weird way.

If you want to capture every events and you do not mind the variable frame rate, you can edit the file `capture.cpp` like so:
```
printf("Sleeping: %d ms\n", asiMaxExposure - (int)(autoExp/1000) + useDelay);
usleep((asiMaxExposure*1000 - autoExp) + useDelay*1000);
 ```
replaced by:
```
printf("Sleeping: %d ms\n", useDelay);
usleep(useDelay*1000);
 ```
Then you'll need to recompile `capture` by running  `make capture` in the terminal

EricClaeys, 2021-09-14:
This issue is from an older release of the AllSky software and a suggested fix supplied.  If you think it might still be relevant, please test it with the newest version and submit a new issue if needed.  Thanks

closed 2021-09-14 by EricClaeys