AllskyTeam/allsky #95: Raspberry 4 and Raspbian Buster : Allsky does not work with the new OS

ZebrasHaveBigEars, 2019-06-25:
Just a note that with the release of the much more powerful [Raspberry 4](https://www.raspberrypi.org/blog/raspberry-pi-4-on-sale-now-from-35/)  this project is going to really fly given its great processing power and USB 3 connections. I have just ordered one. Woohoo!

On the other hand, I have had to reinstall everything once again on my Pi 3B due to a corrupted card and as far as I can see the present install of allsky is now incompatible with the updated operating system: Twice now after install of allsky the new OS does not boot. :-(

thomasjacquin, 2019-06-27:
Thanks for the heads up, I will try it at home and see what's breaking.

richiejarvis, 2019-06-27:
Same report here - this is the error I see:

```
pi@pi4:~/allsky $ sudo ./install.sh

**********************************************
*** Welcome to the Allsky Camera installer ***
**********************************************

* Sunwait installation
* Dependencies installation
Hit:1 http://raspbian.raspberrypi.org/raspbian buster InRelease
Hit:2 http://archive.raspberrypi.org/debian buster InRelease
Reading package lists... Done
Reading package lists... Done
Building dependency tree
Reading state information... Done
Package libav-tools is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source
However the following packages replace it:
  ffmpeg

E: Package 'libav-tools' has no installation candidate

* Using the camera without root access

* Autostart script
sed: can't read /home/pi/.config/lxsession/LXDE-pi/autostart: No such file or directory

* Compile allsky software
arm-linux-gnueabihf-g++  capture.cpp lib/armv7/libASICamera2.a -o capture -Wall -Wno-psabi -g  -I  -L  -D_LIN -D_DEBUG   -L../lib/armv7 -I../include  -lpthread  -I libusb/ -L libusb/   -DGLIBC_20 -march=armv7 -mthumb -I/usr/include/opencv -lopencv_shape -lopencv_stitching -lopencv_superres -lopencv_videostab -lopencv_aruco -lopencv_bgsegm -lopencv_bioinspired -lopencv_ccalib -lopencv_datasets -lopencv_dpm -lopencv_face -lopencv_freetype -lopencv_fuzzy -lopencv_hdf -lopencv_line_descriptor -lopencv_optflow -lopencv_video -lopencv_plot -lopencv_reg -lopencv_saliency -lopencv_stereo -lopencv_structured_light -lopencv_phase_unwrapping -lopencv_rgbd -lopencv_viz -lopencv_surface_matching -lopencv_text -lopencv_ximgproc -lopencv_calib3d -lopencv_features2d -lopencv_flann -lopencv_xobjdetect -lopencv_objdetect -lopencv_ml -lopencv_xphoto -lopencv_highgui -lopencv_videoio -lopencv_imgcodecs -lopencv_photo -lopencv_imgproc -lopencv_core -lusb-1.0
capture.cpp: In function ‘void cvText(IplImage*, const char*, int, int, double, int, int, int, int*, int)’:
capture.cpp:49:25: error: no matching function for call to ‘cv::Mat::Mat(IplImage*&)’
     cv::Mat mat_img(img );
                         ^
In file included from /usr/include/opencv2/core/mat.hpp:3518,
                 from /usr/include/opencv2/core.hpp:59,
                 from /usr/include/opencv2/core/types_c.h:101,
                 from /usr/include/opencv2/core/core_c.h:48,
                 from /usr/include/opencv/highgui.h:45,
                 from capture.cpp:2:
/usr/include/opencv2/core/mat.inl.hpp:1179:1: note: candidate: ‘cv::Mat::Mat(cv::Mat&&)’
 Mat::Mat(Mat&& m)
 ^~~
/usr/include/opencv2/core/mat.inl.hpp:1179:1: note:   no known conversion for argument 1 from ‘IplImage*’ {aka ‘_IplImage*’} to ‘cv::Mat&&’
In file included from /usr/include/opencv2/core.hpp:59,
                 from /usr/include/opencv2/core/types_c.h:101,
                 from /usr/include/opencv2/core/core_c.h:48,
                 from /usr/include/opencv/highgui.h:45,
                 from capture.cpp:2:
/usr/include/opencv2/core/mat.hpp:977:14: note: candidate: ‘cv::Mat::Mat(const cv::cuda::GpuMat&)’
     explicit Mat(const cuda::GpuMat& m);
              ^~~
/usr/include/opencv2/core/mat.hpp:977:14: note:   no known conversion for argument 1 from ‘IplImage*’ {aka ‘_IplImage*’} to ‘const cv::cuda::GpuMat&’
/usr/include/opencv2/core/mat.hpp:974:37: note: candidate: ‘template<class _Tp> cv::Mat::Mat(const cv::MatCommaInitializer_<_Tp>&)’
     template<typename _Tp> explicit Mat(const MatCommaInitializer_<_Tp>& commaInitializer);
                                     ^~~
/usr/include/opencv2/core/mat.hpp:974:37: note:   template argument deduction/substitution failed:
capture.cpp:49:25: note:   mismatched types ‘const cv::MatCommaInitializer_<_Tp>’ and ‘IplImage*’ {aka ‘_IplImage*’}
     cv::Mat mat_img(img );
                         ^
In file included from /usr/include/opencv2/core.hpp:59,
                 from /usr/include/opencv2/core/types_c.h:101,
                 from /usr/include/opencv2/core/core_c.h:48,
                 from /usr/include/opencv/highgui.h:45,
                 from capture.cpp:2:
/usr/include/opencv2/core/mat.hpp:970:37: note: candidate: ‘template<class _Tp> cv::Mat::Mat(const cv::Point3_<_Tp>&, bool)’
     template<typename _Tp> explicit Mat(const Point3_<_Tp>& pt, bool copyData=true);
                                     ^~~
/usr/include/opencv2/core/mat.hpp:970:37: note:   template argument deduction/substitution failed:
capture.cpp:49:25: note:   mismatched types ‘const cv::Point3_<_Tp>’ and ‘IplImage*’ {aka ‘_IplImage*’}
     cv::Mat mat_img(img );
                         ^
```
It all goes pear-shaped from here: `E: Package 'libav-tools' has no installation candidate`

Checking in the buster repo, it seems there is no libav-tools present: https://packages.debian.org/search?suite=buster&section=all&arch=any&searchon=names&keywords=libav-tools

However, there is ffmpeg, and it is installed on my system.  

Looking forward to getting this working with the new beast, so you can add me to the map :)

Cheers,

Richie Jarvis
https://deepsky.org.uk

thomasjacquin, 2019-06-27:
Thanks, that's great information. Looks like avconv got replaced by ffmpeg in that Raspbian release. Hopefully that's the only thing that needs to be changed.

brianboru82, 2019-06-29:
Yep, I'm having the same issue Richie did while installing on a new Pi 3+.

Having ffmpeg installed doesn't seem to help, I was able to find libav directly, and tried cloning and compiling that on the Pi, but still apparently no luck...

Thanks for looking into it!

kpjamro, 2019-07-06:
Cheering you on. Just got my camera and of course am starting with Buster as the OS. Guess I will stop and try to find an older version somewhere ...

jpaana, 2019-07-06:
I just created a PR for fixing compiling of capture.cpp for Buster.

brianboru82, 2019-07-08:
Thank you for that Jpaana, I tried it out and it compiled successfully for me! 

There were some other config files I had to edit to have a successful installation, I will try to assemble those together in the next week or so.

ZebrasHaveBigEars, 2019-07-14:
I installed on my new Raspberry 4 with 4GB today. It all seems to be functioning well. Tremendous work.  I think the software will work much better with the extra processing speed. I am looking forward to seeing how much more quickly the mp4 is produced compared to the Raspberry 3 which struggled under the load. I really think that this is one of the very best Raspberry Pi Projects out there.

richiejarvis, 2019-07-14:
YAY! amazing news - will give it another go...

ZebrasHaveBigEars, 2019-07-14:
And remember to install and use [dropbox]([https://github.com/andreafabrizi/Dropbox-Uploader])) as well as set SSH to yes so that you can ftp. This makes quite a difference to functionality as well.

Cheers

ZHBE

Fulhair, 2019-07-16:
So would you now say the Pi4 is supported? I was going to buy a Pi specifically for my all sky camera set up and have the choice of a 3b+ or a 4? Whats the best option and are there any specific tricks to get it working on a 4? 
Cheers
Fulhair

ZebrasHaveBigEars, 2019-07-16:
Yes - I have it all working with a Pi 4. I installed Buster, Teamviewer and Dropbox and then took an image of the SD card. In raspi-config I set SSH to Yes so that I could FTP to the raspberry. Then I installed the AllSky software. It all worked fine first time. 

It is not necessary but I did a little tailoring of the files in the scripts folder to make the auto FTP from the raspberry work and to include sending files using the dropbox option.  I also changed saveImageDay  so that photos taken during the day  are saved so that they can be included in the auto-generated MP4 file.  All that remains is meteor analysis .... LOL

EricClaeys, 2021-09-14:
This issue is from an older release of the AllSky software and appears to have been resolved.  If you think it might still be relevant, please test it with the newest version and submit a new issue if needed.  Thanks

closed 2021-09-14 by EricClaeys