AllskyTeam/allsky #98: Export config via UI

markwalkom, 2019-07-06:
It'd be great if you could do this to then take the downloaded config file and store somewhere as a backup.

I realise you can take the config from the local filesystem, but this would be a handy alternative.

thomasjacquin, 2019-07-18:
Thanks for the idea Mark. I agree this would be a nice feature.

EricClaeys, 2021-09-14:
Has any progress been made on this, or should the issue be closed?  It's over 2 years old.

Jonk2, 2021-09-20:
I would actually like to see this progressed - it;s all very well on the instructions saying 'back up your config files' but to the uninitiated like me, this isn't obvious.
Also, I have deleted the allsky folder, re cloned a fresh onem installed it and a lot of my settings remained from the previous install.
At least, I 'think' I remember that happening, can't say for sure.

EricClaeys, 2021-09-20:
@Jonk2,
if you have the GUI installed, the camera settings (the ones on the GUI) are stored in /etc/raspap.  They are not touched when you re-install allsky, although they aren't used unless your config.sh had `CAMERA_SETTINGS=/etc/raspap`, which the default file does not.  Is it possible you put that line into the new config.sh?

They /etc/raspap configuration files ARE overwritten with defaults when you reinstall allsky-portal.

One of the "to-do" items is to have an update script that would, among other things, would prompt the user if they want to save their current settings and restore them after the upgrade.  Combined with the plan to put ALL configuration files in ~/allsky/configuration, and remove /etc/raspap, it would be pretty easy to "back up" your settings - just copy the ~/allsky/configuration

@markwalkom,
I added this feature to the "possible to-do" list that Thomas, Chris, and I can discuss.

Jonk2, 2021-09-20:
To be honest, when I first found this software, it appeared to have gone stale, i.e. before you became involved it hadn't been updated for a long time.
I wouldn't have done something so complex if I didn't have a small understanding of what I was doing, so I probably didn't add anything like that.
Having said that, yes a simple 'save config / settings' button / option would be most welcome.

markwalkom, 2021-09-20:
> To be honest, when I first found this software, it appeared to have gone stale, i.e. before you became involved it hadn't been updated for a long time.

The developers of this work on this as a labour of passion, and they have no obligations to continue to work on or support it moving forward. Please consider supporting the project to show you appreciate their efforts (if you haven't) - https://github.com/thomasjacquin/allsky#donation

EricClaeys, 2021-09-20:
@Jonk2, @markwalkom,
About 6 months ago I got an allsky camera and a Pi and found the AllSky software via Google and downloaded it.  I also noticed it was somewhat stale and I had a lot of questions and issues.  I recently retired and was feeling the urge to program, so I forked the software and started updating my copy.  I also started participating in the Issues on GitHub.

At some point Linda Thomas-Fowler said she was developing a workaround to the ZWO daylight auto exposure bug, so we teamed up.  A couple more people joined, and we created version 0.8 after some conversations with Thomas, who was glad to see the changes.

Recently @ckuethe started participating and we've tossed some improvement ideas back and forth, and he's submitted several changes, so I'm hopeful we'll be able to quicken the pace of updates (but not too quick that we break things).

My 60 second thought on the requested feature is to have a simple form for requesting a configuration backup.  The user would enter the name of a directory they want the backup stored in, would press the "Backup" button, and the configuration files would be copied to that directory.  I suppose a "restore" option would be useful.
Keep in mind, I haven't looked at the feasibility of this, and I wouldn't attempt it until after all the configuration files are moved to a single directory.

ckuethe, 2021-09-20:
My impulsive thought is: get rid of all the shell based configs, and replace them with a single JSON settings file which shall be the authoritative source of all settings. The replace the shell scripts with python scripts which are way faster (~100x) for parsing and computing command line arguments from JSON.

JSON has the advantage of being machine-readable, machine writable, and fairly easy to check for being correctly formatted.

Something else worth considering in exporting configuration is any user-modified scripts, eg. ftp settings or whatever else is done for nightly uploads.

EricClaeys, 2021-09-20:
@ckuethe, The settings files I'm aware of are:
- camera_options_*.json
- settings_*.json
- config.sh
- ftp_settings.sh
- autocam.sh

And the repo files:  settings_*.json.repo,  config.sh.repo, and ftp_settings.sh.repo.

Some desired future settings include camera capabilities' min, max, and default values as well as camera model and ID (where available).

Whatever the design is needs to allow for multiple cameras with their own settings (e.g., crop), as well as common settings (e.g., latitude).  Or maybe there aren't any common settings and everything is camera-specific.

ckuethe, 2021-09-20:
The repo files should not be backed up. They come from git, and should never be user-modified. Their purpose is to give and example of how to configure that stage of processing. 

IMHO, the existence of autocam.sh is a  major design flaw. When I saw it I cursed a lot and then commented it out. If I put `CAMERA="ZWO"` in my config.sh, allsky has no business running any autodetection - clearly I had a preference, so just respect it. If it doesn't work - that's on me to fix. Additionally, the autodetection shouldn't be run on non-arm machines.
https://github.com/thomasjacquin/allsky/blob/master/allsky.sh#L8-L10

I also assert that putting settings in /var/www/html is wrong. What I'd prefer to see is all the settings for all things allsky be in the allsky directory (or at least one common place) and that a symlink or a bind mount be used to make the GUI use the same settings.

How do you propose to differentiate camera settings if I have two different ZWO cameras (eg. a 120 and a 183)? What if I have two of the same cameras (eg. two ASI120MM-mini)? You can't assume that I have the same lens, FoV, filters, or pointing direction on each. Do we have any examples of multi-camera allsky installations?

It occurs to me that we can embed other files into the body of the backup file, in a couple of ways:
* encode the file in base64, use that as the value, and the key is the filename
* encode the file as an array of strings, use that as the value, and the key is the filename.

EricClaeys, 2021-10-16:
@ckuethe All the WebUI's settings are either in allsky/config or /etc/raspap.  Once we implement a single config file they'll all be in that file.
`allsky-website` has a couple configuration files in `/var/www/html/allsky` if it's on the Pi, or some other directory if it's on a different computer.  We could put those settings in the big, single file, and have `allsky-website` check if it's on a Pi, it would use that file, otherwise it would use the existing file.

EricClaeys, 2022-05-18:
Closing this Issue.  This enhancement is on our "to do" list.

closed 2022-05-18 by EricClaeys