AllskyTeam/allsky #100: Raspberry Pi 4

spacemanbobby, 2019-07-17:
I just tried for fun.. Got the new Raspberry Pi 4. Installed OS on it then the project..  runs but the camera doesn't work. Dang it.   

Any idea if there is a chance you will compile for this new version of board. It is so much faster than the raspberry 3+

dbainbridge, 2019-07-17:
This should be working see other issue here: https://github.com/thomasjacquin/allsky/issues/95#issuecomment-511999630

kpjamro, 2019-07-17:
I built one on Buster yesterday and have been playing with it (I am new to this).  The camera (capture) works and makes png files. Lots of png files that overflowed my SD card (images are 10MB each with my camera in RAW 16 mode).  I deleted enough files to have about 50% of the memory free and a rebooted the pi.

Under this condition two things don't seem to work.  

One is the mp4 generation (timelapse.sh). It results in a file that is empty except for a little bit of header information.  There are no error messages as such, but I see:
./timelapse.sh: line 29:  4248 Killed                  ffmpeg -y -f image2  (etc)
and I suspect that is not normal? In any case, after running timelapse in a console window, command line text is no longer visible when I type.

The other thing that did not work is the "public" image display.  I get just a black screen with some indication on the browser that it is waiting .....

spacemanbobby, 2019-07-17:
I just saw camera information displayed on the top left even though I
didn't have a camera connected. Just showed a black screen. I'll try
changing the images to png and let you know what happens.

kpjamro, 2019-07-17:
Camera info was showing in my "live view" under JPG, but but not PNG.  RAW 16 requires the use of PNG (according to the menu).  You *can* select JPG with RAW 16, but then auto-exposure goes crazy.

kpjamro, 2019-07-17:
BTW - you can watch what the program is doing by running:
tail -f /home/pi/allsky/log.txt
in a console window.

kpjamro, 2019-07-17:
> Camera info was showing in my "live view" under JPG, but but not PNG. RAW 16 requires the use of PNG (according to the menu). You _can_ select JPG with RAW 16, but then auto-exposure goes crazy.

Actually - now that I am looking at it, PNG does the same thing.  It just runs the exposure up to the maximum allowed exposure time and pegs it there.  I was just lucky that the value was a reasonable one to begin with .

kpjamro, 2019-07-18:
I probably should be posting this somewhere else but let me summarize.
I have a monochrome camera (178MM) and using RAW16/PNG.  
Unexpected observations are:
> text overlay does not work
> metadata is always in black, but responds to the font size setting
> time-between does not work. The time between is always the exposure length plus a fraction
> image quality for PNG is actually compression ratio, so 9 is max compression not max quality.

I am doing 4-bin tonight in hopes that it fits on the SD card. 
We'll see if there is a movie tomorrow AM :-)

kpjamro, 2019-07-18:
Yay - I got the movie. It seems - left unmolested - the movie (etc) creation works fine in the Buster version of Allsky - at least with the smaller files.  My next step will be to remap the images to a thumb-drive instead of the SD card so I can accommodate the full-sized images.

I can understand why the non-functionality of "time-between" has not  been noticed - people normally don't want to miss a minute of meteor activity  ;-)

It will be a couple of weeks before I actually get my RPI-4 ... FLIRC say they will start shipping cases at the end of July. This will give me time to build the enclosure and mounting.

spacemanbobby, 2019-07-21:
Where is the Buster version of allsky

spacemanbobby, 2019-07-21:
Actually I have the Buster version...  This is what I am running now and
the ASI185 camera is not previewing images or saving images.. Just a black
screen in preview that shows data even if I do not connect my camrea.

Any Hints..

Raspbian Buster with desktop and recommended software
Image with desktop and recommended software based on Debian Buster
Version:July 2019
Release date:2019-07-10
Kernel version:4.19
Size:1945 MB

kpjamro, 2019-07-21:
If the program is not throwing off error messages about openCV, then you must have the Buster version.

Does the status indicate the program is running OK?
sudo service allsky status

Did the camera work ok with the OEM software?

One thing I did notice is that when you first start the camera with Allsky it seems to use 32 microsecond exposures and slowly increases exposure length.  It can be some minutes before you see anything in the preview window.  I am using an ASI178MM.

kpjamro, 2019-07-21:
pi@allsky:~ $ sudo service allsky status
â allsky.service - All Sky Camera
   Loaded: loaded (/lib/systemd/system/allsky.service; enabled; vendor preset: enabled)
   Active: active (running) since Sun 2019-07-21 13:19:39 EDT; 40s ago
 Main PID: 22852 (allsky.sh)
    Tasks: 7 (limit: 2200)
   Memory: 176.3M
   CGroup: /system.slice/allsky.service
           ââ22852 /bin/bash /home/pi/allsky/allsky.sh
           ââ22901 ./capture -angle -6 -autoexposure 0 -autogain 1 -bin 1 -brightness 50 -darkframe 0 [etc]

Jul 21 13:19:39 allsky systemd[1]: Started All Sky Camera.
Jul 21 13:19:39 allsky allsky.sh[22852]: can't get debug descriptor: Resource temporarily unavailable
lines 1-12/12 (END)

spacemanbobby, 2019-07-21:
The Allsky Program looks like it is running.. I'll set it up again tonight
and see what happens.

ZebrasHaveBigEars, 2019-07-22:
I can confirm that my installation is working fine with Raspberry 4 and Buster.

kpjamro, 2019-07-22:
can I ask how much memory in the RPI4 (1,2,4GB)? and at what resolution is the camera is being read?

spacemanbobby, 2019-07-22:
2GB version of pi 4.

185mc isn't super high resolution. Was a 350 dollar camera.
Resolution: 2.3 Mega Pixels 1944*1224

Has been working fine on my other older pi 3+ system

kpjamro, 2019-07-22:
OK - my question was directed at Zebra - but I guess image size is not your problem.  With my RPI3B I found that there was not enough memory to run the FFMPEG conversion with my PNG files that were about 10M each (but I could make the AVI offline on another machine or I could use binning to make the file a lot smaller).  

Next I tried different settings for image quality using a rather plain scene that actually could be compressed (a starry sky seems not very compressible). 
setting 5 -> 5.3 MB
setting 2 -> 8.3 MB
setting 0 -> 12.6 MB (no compression)
after transferring the 12.6 PNG the program lost its ability to read data from the camera - either settings or the picture.  I assume the SD card got corrupted. The camera still works with other software.

RPI-4 came today! so I am just moving forward with that, starting with 2GB size.

kpjamro, 2019-07-27:
OK - so the 2GB PI4 works.  It pumps the CPU so hard that it goes into temperature limit (throttling) when making the movie.  But I do not have a heatsink on it yet. It makes 3 fps until limiting kicks in then it drops to about 1.7 fps
memory - 82%
CPU - 180%
Temp - 82 degrees C

Only problem outstanding now is that star track produces an entirely white sky image. I tried different threshold values without luck.

spacemanbobby, 2019-07-29:
I reinstalled buster on the system. It saw my camera and was doing live
updates. As soon as I changed settings in the config and scripts it stopped
updating the camera image.

Did this twice just to make sure. Each time I make changes to files so it
is like my old pi3+ (camera settings) it just stops updating.

I ran the code to see status and it says it is online but it won't update
the image anymore.

Any ideas?

kpjamro, 2019-07-29:
Yes - exactly what I saw.  Sometimes I was able to restart by rebooting (but not always).  It works fine at night.  Try changing your longitude so that it "thinks" it is night time when/where you are and I will not be surprised if it starts behaving.

spacemanbobby, 2019-07-30:
I'll give it a shot but in all reality, I have my webpage setup to show a
live view24/7 if you go to the page currently with pi 3+.

One thing.. Are the images any better or are they the same with this new
version of buster and the hardware on the pi board

kpjamro, 2019-07-30:
My motivation to go to the PI4  is that the PI3 simply cannot handle my 6 Megapixel camera unless I bin it down to a lower resolution.  So in my case, yes, the image is better  ;-)

kpjamro, 2019-07-30:
I found that -
If you stop Allsky using the menu, then SSH into the PI and start the program using: 

./allsky.sh 

The you can make camera setting changes (using the GUI) - during the daytime - and the program does not stop running.  This could be a hint as to where the problem (bug) is. 

Unfortunately, you have to *leave* that window open...

EricClaeys, 2021-09-14:
This issue is from an older release of the AllSky software.  If you think it might still be relevant, please test it with the newest version and submit a new issue if needed.  Thanks

closed 2021-09-14 by EricClaeys