AllskyTeam/allsky #101: Status with error

WirthmU, 2019-07-18:
I am trying to install allksy on a RPI B+.

If I check the status I get:

`pi@SkyCam:~/allsky/scripts $ sudo service allsky status
● allsky.service - All Sky Camera
   Loaded: loaded (/lib/systemd/system/allsky.service; enabled; vendor preset: enabled)
   Active: active (running) since Thu 2019-07-18 21:18:28 BST; 12s ago
 Main PID: 1136 (allsky.sh)
   Memory: 88.9M
   CGroup: /system.slice/allsky.service
           ├─1136 /bin/bash /home/pi/allsky/allsky.sh
           └─1150 ./capture -daytime 1

Jul 18 21:18:28 SkyCam systemd[1]: Started All Sky Camera.
Jul 18 21:18:28 SkyCam allsky.sh[1136]: can't get debug descriptor: Resource temporarily unavailable
Jul 18 21:18:28 SkyCam allsky.sh[1136]: parse error: Invalid numeric literal at line 2, column 5
Jul 18 21:18:28 SkyCam allsky.sh[1136]: parse error: Invalid numeric literal at line 2, column 5`

What might be the problem here?

brianboru82, 2019-07-24:
For the longitude/latitude, I believe you do need the N and E (or S and W as appropriate).

Not sure what the parse errors are about...

thomasjacquin, 2019-08-20:
Did you get this sorted?

WirthmU, 2019-08-21:
Hello Thomas and thank you for getting back to me!

Unfortunately no - I am still struggling with the errors stated above. Here are my settings and config files:

`settings.json:``

{
   "width":"0",
   "height":"0",
   "exposure":"10000",
   "maxexposure":"20000",
   "autoexposure":"1",
   "gain":"50",
   "maxgain":"200",
   "autogain":"0",
   "gamma":"50",
   "brightness":"50",
   "wbr":"53",
   "wbb":"90",
   "bin":"1",
   "delay":"10",
   "daytimeDelay": "5000",
   "type":"1",
   "quality":"95",
   "usb":"40",
   "filename":"image.jpg",
   "flip":"0",
   "text":"text",
   "textx":"15",
   "texty":"30",
   "fontname":"0",
   "fontcolor":"255 255 255",
   "smallfontcolor":"0 0 255",
   "fonttype":"0",
   "fontsize":"0.7",
   "fontline":"1",
   "latitude":"46.94N",
   "longitude":"7.475E",
   "angle":"-6",
   "time":"1",
   "darkframe":"0",
   "showDetails":"1"
}

config.sh:

#!/bin/bash
source /home/pi/allsky/scripts/ftp-settings.sh

# Set to true to upload current image to your website
UPLOAD_IMG=true

# Set to true to upload timelapse video to your website at the end of each night
UPLOAD_VIDEO=true

# Set to true to send data to your server at the end of each night
POST_END_OF_NIGHT_DATA=true

# Set to true to generate a timelapse at the end of the night
TIMELAPSE=true

# Set to true to generate a keogram at the end of the night (image summary of the nigh$
KEOGRAM=true
# Set to true to upload the keogram after it has been generated
UPLOAD_KEOGRAM=true

# Set to true to generate a startrails image of the night. Will skip brighter images t$
STARTRAILS=true

# Images with a brightness higher than the threshold will be skipped for startrails im$
BRIGHTNESS_THRESHOLD=0.1

# Set to true to upload the startrails after it has been generated
UPLOAD_STARTRAILS=true

# Set to true to enable automatic deletion of archived data (images + videos)
AUTO_DELETE=true

# Set this value to the number of archived nights you want to keep. Needs AUTO_DELETE=$
NIGHTS_TO_KEEP=14

# Path to the dark frame for hot pixels subtraction. Can be jpg or png.
DARK_FRAME="dark.png"

# Set to 0 to disable Daytime Capture
DAYTIME="1"

# Path to the camera settings (exposure, gain, delay, overlay, etc)
CAMERA_SETTINGS="/home/pi/allsky/settings.json"`

WirthmU, 2019-08-23:
After a complete removal, updating and upgrading my RPI and re-installation its working!

closed 2019-08-23 by WirthmU