AllskyTeam/allsky #102: allsky.sh (code exited status 127)

RobertG73, 2019-07-20:
Hi! I just started this project, installed all, connected an ASI 120MC camera but when i launch "status" command i got the erros from the topic title.
I have a Raspberry PI 3 B+, and i hope i tried to not miss any step from instalation tutorial.
i can wait to start this project and thanks for sharing this!

brianboru82, 2019-07-24:
It sounds like the install didn't work exactly right. 

- For the jq, in terminal, type `sudo apt-get install jq` and that should install it.

- For the line 32, it's referencing the compiled capture program. It sounds like it didn't compile properly, so it doesn't exist. Try running the install.sh again, and see what errors it shows during the install process.

RobertG73, 2019-07-24:
Thanks for the answer!
I deleted the folder "allsky", manually installed "jq", this time, this command is working if i tipped un console.
Reinstalled the "allsky" program step by step.
Still got the error "Line 32 ./capture no such file or directory"
As a PI system i have raspbian Buster desktop.

RobertG73, 2019-07-25:
Update:
Yesterday it was late, but luckly i made a picture of the allsky installation process.
Looking at the picture now, i see that "opencv" package is not found in pkg-config path.
I am looking out how to fix this.

brianboru82, 2019-07-25:
It sounds like your copy of the allsky install files is out of date; there were updates made a few weeks ago that should fix that. I would try doing the 'git clone' process again and downloading a fresh copy, and try again. (OpenCV has been replaced by ffmpeg in the most recent version of Raspbian).

RobertG73, 2019-07-25:
This is what i made yesterday, cloned the project (git clone https://github.com/thomasjacquin/allsky.git)from here on my PI and i do not understand for the moment why isn't working. and i still have problems.

RobertG73, 2019-07-28:
I was takel all from zero, installed open CV, ffmpeg, and allsky software... still the same problem.
Where i am wrong?
i attach a picture with the error, and the log content.
Sometimes, but very rarely, the status command looks ok, the software running, but no images are generated, and if i run status again i found the allsky software exited with code 127...
[image]

thomasjacquin, 2019-07-28:
Hi Robert, you do not need sudo to run ./allsky.sh

RobertG73, 2019-07-28:
Hi, i just started manually to see what are the errors, and why the capture is int working

thomasjacquin, 2019-07-28:
It looks like you don’t have the file called ‘capture’. It should be compiled during the install script. Try re-running the install script and look for errors in the terminal. There might be something that failed to install.

RobertG73, 2019-07-28:
Ok, i have "capture.cpp" in the allsky folder. I will delete /allsky ant rerun instal script and look for any errors.
I edited my post from above with more details :)

RobertG73, 2019-07-28:
[image]
Looks like i still have problems with "openCV" even i i made all the steps installing it...

brianboru82, 2019-07-30:
For that first error at the very top, type `sudo apt update -y` in terminal to update it.
Try running 'sudo apt-get install libopencv-dev' in terminal to see if it can install OpenCV that way, and then run the allsky install.sh?

RobertG73, 2019-08-04:
Thank you for reply! F
[image]
or the first time looks like OpenCV is corectly installed. I i run the command "pkg-config opencv --cflags" it returns "-I/usr/include/opencv"
I still have an error when i reinstall AllSky software, i attach the picture with the installation process.

ChrisFromDE, 2019-08-20:
Hi, a have a similar issue:
when starting the gui (without all is ok) i get the error in line  32
./capture $ARGUMENTS

above this i see an error in the opencv (i'm not at my Raspbian). So i have to built the bin again or have i to update the opencv libraries?
My Raspbi3b+ and Raspbian is up to date, all packages are up to date, too (i hope) .
sudo apt-get update
sudo apt-get upgrade

i'm not firm with Linux and i don't know what to do, so any help is welcome

Regards from Germany
Chris

RobertG73, 2019-08-24:
I think you need to recompile openCV. With help from one colleague (he made to me a list with all instructions to do step by step) a noob like me managed to correctly install openCV. I attach here all the steps, maybe it will help somebody else...
[opencv-install.txt](https://github.com/thomasjacquin/allsky/files/3537059/opencv-install.txt)

closed 2019-08-27 by RobertG73