AllskyTeam/allsky #104: question: is it possible to remap the file storage and working folder

mth75, 2019-07-23:
Is it possible to remap the file storage and working folder say on a USB thumb drive? If yes, how to. Best regards!

kpjamro, 2019-07-30:
Jusst mount your thumb (or any) drive per instructions you will find on the internet giving it the path name 

/home/pi/allsky/images

and the program will start using it.  Any images in the old location (SD card) will no longer be accessible, so save them out first if you need them.  Here is the result of "df" on my system with a 500 GB mini SSD being used in this way

Filesystem     1K-blocks     Used Available Use% Mounted on
/dev/root       61058964  6175568  52323700  11% /
devtmpfs          860916        0    860916   0% /dev
tmpfs             993012        0    993012   0% /dev/shm
tmpfs             993012     8676    984336   1% /run
tmpfs               5120        4      5116   1% /run/lock
tmpfs             993012        0    993012   0% /sys/fs/cgroup
**/dev/sda1      488384000 22490496 465893504   5% /home/pi/allsky/images**
/dev/mmcblk0p1    258096    39970    218126  16% /boot
tmpfs             198600        0    198600   0% /run/user/1000

mth75, 2019-07-30:
thx for your answer!

ckuethe, 2019-09-20:
Revisiting this, I'd like to remap the capture directory to `/tmp` or some other memory filesytem so I don't trash my sdcard writing and writing and writing and writing the same file over and over. For now I'm doing some ugly things with tmpfs and overlayfs to allow the live images to just live in memory.

kpjamro, 2019-09-20:
see  https://github.com/thomasjacquin/allsky/issues/104#issuecomment-516545079

ckuethe, 2019-09-20:
Nope. That doesn't help in preventing writes to the current directory. I'm already storing the images somewhere else, but `image.jpg`, `image-processed.jpg`, and `liveview-image.jpg` are still written to the allsky directory.

kpjamro, 2019-09-20:
Oh - I see what you mean. Within a compiled program I don't think working files actually get written to non-volatile storage until the program ends, or a "sync" instruction is executed.  But in a script, I imagine they get written at the end of each line.  I think what would be needed is to modify the scripts that write these files so that, instead of the filename presently used, it uses /tmp/filename.

Just changing the file name in the camera settings doesn't quite do it, because of the way he manipulates the file names.  But it shouldn't be too hard to do.... maybe I will give it a shot tonight.

ckuethe, 2019-09-20:
[capture.cpp:117](https://github.com/thomasjacquin/allsky/blob/master/capture.cpp#L117) is going to cause writes to non-volatile storage, which will then be picked up by [saveImageNight.sh](https://github.com/thomasjacquin/allsky/blob/master/scripts/saveImageNight.sh). Maybe that will get synced to storage sooner rather than later, maybe not, depending on other workloads on the device. One could use `btrace` or `blktrace` to see.

Way down my list of priorities is adding a configurable knob for `cap_dir`, `tmp_dir`, or something like that to send each frame grab to volatile storage before it gets processed. I'm working on some other stuff with INDI right now so I won't be sad if someone gets around to implementing this before I do.

kpjamro, 2019-09-20:
If the filename variable is "/tmp/image.png" or similar then capture will write it to the RAM memory only.  The problem is that elsewhere in some of the scripts the /tmp/ at the start of the filename will screw things up....

ckuethe, 2019-09-20:
Yes, I found that the hard way yesterday. Thus the configuration knob for a work directory so that all the scripts, cgis, binaries, ... agree on where to write.

EricClaeys, 2021-09-14:
@ckuethe, can this issue be closed?  It's a couple years old and the original question was answered.  With Thomas' approval, I'm trying to close old issues so we have a list of current ones that need attention.

I'm looking at adding a $ALLSKY_TMP variable that would point to a location to store all temporary files like the log files and the image created by capture.cpp which is then optionally edited by all the scripts.  The final image used by the web sites could also reside in $ALLSKY_TMP, as well as the temporary timelapse sequence/* files.  Users wanting to save on SD card writes could map $ALLSKY_TMP to a tmpfs filesystem or a USB drive.

EricClaeys, 2021-09-14:
Chris, sorry, I should have been clearer when I said "log files".  I meant log files created by endOfNight.sh for keograms, startrails, and timelapse output that used to clutter up /var/log/allsky.log, and are overwritten every night so don't grow endlessly.  They are used for debugging purposes.

EricClaeys, 2021-10-16:
The `allsky/variables.sh` file sets a variable called `ALLSKY_TMP` that by default is `allsky/tmp` but could be changed to point to a directory on a memory filesystem.  A future version of allsky will write the "image.jpg" file to $ALLSKY_TMP, and do all processing there, only copying the final version to $ALLSKY_IMAGES/date.  If $ALLSKY_TMP is a memory filesystem this will SIGNIFICANTLY reduce the activity on the SD card.

EricClaeys, 2022-02-10:
Closing.  Version 0.8.3 saves all images from the camera to allsky/tmp, and does all processing there.  The Wiki has a page that describes how to make allsky/tmp a `tmpfs` filesystem which will significantly reduce SDcard writes.

closed 2022-02-10 by EricClaeys