AllskyTeam/allsky #105: Pi 4 newest rasbian buster cannot access allsky program WAN

wannaberocker, 2019-07-31:
Hello, I cannot get to my allsky via WAN, locally it works ( except for the public image is broken ).

Did something change about raspbian that is causing me not being able to access my allsky via port forwarding? My old one on Pi 3 works fine ( accept for broken public image ) and my Pi 4 is set up exactly the same but I cannot get to it from WAN.

mth75, 2019-08-03:
Confirmed, the same problem on my Pi4 (with Raspbian Buster Lite)

thomasjacquin, 2019-08-20:
I should receive a Pi 4 unit in a few days and I will investigate the cause of this.

kpjamro, 2019-11-27:
I have been running RPI-4 locally without problems and recently (for various reasons) have put the Allsky behind a router, assuming I can reach it from the WAN side via port forwarding.  But no. I did not realize this was a known issue.  I spend many hours trying to figure out what I was doing wrong with the forwarding (!)

Anyway - has this been figured out?  The Allsky RPI-4 is a static IP (/etc/dhcpcd.conf) running Buster of course.

I am forwarding port 80 both TCP and UDP in the router.  I even tried to forward port 22 for SSH.  That does not work either. 

Help.

kpjamro, 2019-12-06:
OK - made some progress on this issue.  It seems that on my system you cannot simultaneously use the wired and wireless connection.  If I turn off or disconnect the wireless link, then my router starts working fine.  I should point out that the router in question is a 2nd router, not the ISP router that supplies the wireless access point.  I believe that somewhere in the system there is some loop detection feature that is blocking the port forwarding action in the router.  With wireless off I can port forward SSH/putty to AllSky and reach other raspberry pi's on that side of the router.

However, I am not having any response to port 80.  I would like to reach the AllSky GUI through the router.  What port is the web server listening to??

thomasjacquin, 2019-12-06:
The GUI is using lighttpd and the default port is 80. You can customize that in lighttpd.conf line 14
https://github.com/thomasjacquin/allsky/blob/master/gui/lighttpd.conf#L14

kpjamro, 2019-12-07:
Thanks

 

I edited the file /etc/lighttpd/lighttpd.conf  and changed the server port to 8080 then rebooted the pi.   Now I can see it’s listening on 8080 TCP.  From the local lan I can point my browser to (lanip):8080 and bring up the GUI page.  I chose a number different than 80 just to see if there was a conflict somewhere.

 

But my browser still will not connect through the router’s port-forwarding (8080->8080).  This is strange since I can connect with SFTP through the very same physical path (in that case port 2230 --> 22).   And I can see some traffic passing from the browser through the router (from the built-in counters in the router). But it will not establish a connection. 

 

I guess this is a router problem, but I wonder why lighttpd does not want to talk through the router…?  I am using the standard firewall rules. 

 

Ken

 

https://github.com/thomasjacquin/allsky/blob/master/gui/lighttpd.conf#L14

kpjamro, 2019-12-13:
I have tried yet another port number without luck.  As previously mentioned I can reach port 22 for the SSH server and also 5900 for the VNC server successfully. (As before, the computer doing the access can be connected by wired or wireless but not both at same time).

But the basic problem is that http server (lighttpd) does not work through the router.  I am forwarding a single TCP port - the one that lighttpd is listening on.  Curious if wannaberocker or mth75 ever got this working?  Or has anybody got this working (with Buster).

EricClaeys, 2021-09-14:
This issue is from a much older release of the AllSky software.  If you think it might still be relevant, please test it with the newest version and submit a new issue if needed.  Thanks

closed 2021-09-14 by EricClaeys