AllskyTeam/allsky #107: Timelapse script process

brianboru82, 2019-08-07:
I was curious about the image renaming process in the timelapse.sh script, where it renames each image from 0001.jpg to xxxx.jpg. Is this necessary for the video processing to sequence the images correctly?

My concern is that this removes the timestamps from the image filename (and updates the file creation/modified date), so it can make it more difficult to go find an image from a certain time of night.

I was thinking of saving an image from the same time each night (say midnight), and then after a year, compile those images into a video to show the precession of the stars relative to the earth.

thomasjacquin, 2019-08-07:
Hey!
I like the idea of making a video using one image per night.
The reason why images are renamed to a sequence is because avconv/ffmpeg required a filename with a number between 0 and 9999 in order to generate a video output.

I would suggest making a copy of the image captured around midnight before the timelapse starts and store it in a ‘precession’ directory.

That way you only need to keep one image per night and that won’t fill up the SD card.

cbassa, 2019-08-07:
Why not make softlinks in `timelapse.sh` instead of copies of the images. That saves on disk space.

thomasjacquin, 2019-08-07:
Well that’s a pretty good idea.
1 Make soft links on all images in the directory
2 Run avconv
3 Remove soft links
4 Find image close to midnight and store it somewhere else because archives are only stored for a certain number of nights (set in config.sh)

brianboru82, 2019-08-09:
Ah, that makes sense. I'll try to figure out the softlinks scripting and image finding as I have time. It might take me a while as I'm not a Linux expert. ;)

thomasjacquin, 2019-08-22:
I pushed a change inspired by cbassa's pull request to address the issue. #4 in the list would need to be treated separately.

brianboru82, 2019-08-24:
Thanks Thomas! Still working on #4 slowly in my free time.

brianboru82, 2019-08-24:
This is what I have so far:
`find /home/pi/allsky/images/20190817/ -maxdepth 1 -type f -newermt "2019-08-18 00:00:00" \! -newermt "2019-08-18 00:00:45"  | sort -tr |head -1 | xargs -I {} cp -n {} /home/pi/allsky/Precession/ `

It finds files in a specific directory, and copies the first file between the two times over to the Precession directory. I think my next step is to make it compatible with a script. It might also be made simpler by using the `-newermt -1` or something similar to find things made in the last 24 hours, which is when this script would most logically be running. Anyway, I'll get to that later.

hdiessner, 2019-08-28:
Hi, great piece of work!
I'd like to bring the topic of soft links together with using an external / mounted file system for storing the images: I am using a mounted folder from my NAS, but CIFS apparently doesn't support symbolic links. So I changed the directory for the soft links to a temporary directory (/home/pi/allsky/temp), have all symlinks created in this directory and after creating the video and saving it to the NAS, the local temp folder is deleted.

So this is "my" timelapse.sh:

`#!/bin/bash
source /home/pi/allsky/config.sh
source /home/pi/allsky/scripts/filename.sh

cd /home/pi/allsky/

RED='\033[0;31m'
GREEN='\033[0;32m'
NC='\033[0m' # No Color

if [ $# -lt 1 ]
  then
    echo -en "${RED}You need to pass a day argument\n"
        echo -en "    ex: timelapse.sh 20180119${NC}\n"
        exit 3
fi

echo -en "* ${GREEN}Creating symlinks to generate timelapse${NC}\n"
mkdir /home/pi/allsky/temp/sequence/
# find images, make symlinks sequentially and start avconv to build mp4; upload mp4 and move directory
find "/home/pi/allsky/images/$1" -name "*.$EXTENSION" -size 0 -delete
ls -rt /home/pi/allsky/images/$1/*.$EXTENSION |
gawk 'BEGIN{ a=1 }{ printf "ln -sv %s /home/pi/allsky/temp/sequence/%04d.'$EXTENSION'\n", $0, a++ }' |
bash
ffmpeg -y -f image2 -r 25 -i temp/sequence/%04d.$EXTENSION -vcodec libx264 -b:v 2000k -pix_fmt yuv420p images/$1/allsky-$1.mp4

if [ "$UPLOAD_VIDEO" = true ] ; then
        lftp "$PROTOCOL"://"$USER":"$PASSWORD"@"$HOST":"$MP4DIR" -e "set net:max-retries 1; put images/$1/allsky-$1.mp4; bye"
fi

echo -en "* ${GREEN}Deleting sequence${NC}\n"
rm -rf /home/pi/allsky/temp/sequence

echo -en "* ${GREEN}Timelapse was created${NC}\n"`

brianboru82, 2019-11-05:
I finally got around to finishing the script for grabbing the midnight photos every night:
It searches for the first photo taken between midnight and 45 seconds after midnight, and then copies it to a Precession folder in the images directory. On my own installation, I added this to the EndOfNight.sh script. Feel free to include if you like it.

`#!/bin/bash
source /home/pi/allsky/config.sh
source /home/pi/allsky/scripts/filename.sh

cd  /home/pi/allsky/scripts
LAST_NIGHT=$(date -d '24 hours ago' +'%Y%m%d')
MIDNT=$(date -d '12am today' +'%c')
MIDNT2=$(date -d '12am today+45 seconds' +'%c')
THIS_DAY=$(date -d 'today' +'%Y%m%d')

# Copy image just after midnight to Precession folder
if [[ $PRECESSION == "true" ]]; then
	echo -e "Finding first image after Midnight for precession\n"
	mkdir -p /home/pi/allsky/images/Precession/
	find /home/pi/allsky/images/$LAST_NIGHT/ -maxdepth 1 -type f -newermt "$MIDNT" \! -newermt "$MIDNT2" | sort -tr |	head -1 | xargs -I {} cp -n -p {} /home/pi/allsky/images/Precession/$THIS_DAY.jpg 
	if [[ $UPLOAD_PRECESSION == "true" ]] ; then
		OUTPUT="/home/pi/allsky/images/Precession/$THIS_DAY.jpg"
#"find /home/pi/allsky/images/Precession/ -maxdepth 1 -type f -newermt "$MIDNT""
                lftp "$PROTOCOL"://"$USER":"$PASSWORD"@"$HOST":"$PRECESSION_DIR" \
                        -e "set net:max-retries 1; put $OUTPUT; bye"
	fi
        echo -e "\n"
fi`

EricClaeys, 2021-09-14:
This issue appears to have been resolved (thanks Brian).  If you think it might still be relevant, please test it with the newest version and submit a new issue if needed.  Thanks

closed 2021-09-14 by EricClaeys