AllskyTeam/allsky #110: strange GUI display

theojurriens, 2019-08-16:
[image]

Looks like a Phone interface on a PC?? And dark mode is unavailable.

Any hints?

Thanks

AnWeNet, 2019-08-19:
theojurriens, 
did you find a solution. I am having the same issue.
it worked fine for last 6 months until I upgraded few days ago.

[image]

theojurriens, 2019-08-19:
same in my case: after update :-(

thomasjacquin, 2019-08-19:
Is there any errors showing up in the developer tools? (right click --> inspect anywhere on the page)

AnWeNet, 2019-08-19:
I only see the following:

[Error] Did not parse stylesheet at 'http://allsky.awcloud.ca:8080/bower_components/bootstrap/dist/css/bootstrap.min.css' because non CSS MIME types are not allowed in strict mode.
[Error] Did not parse stylesheet at 'http://allsky.awcloud.ca:8080/bower_components/metisMenu/dist/metisMenu.min.css' because non CSS MIME types are not allowed in strict mode.
[Error] Did not parse stylesheet at 'http://allsky.awcloud.ca:8080/dist/css/timeline.css' because non CSS MIME types are not allowed in strict mode.
[Error] Did not parse stylesheet at 'http://allsky.awcloud.ca:8080/dist/css/sb-admin-2.css' because non CSS MIME types are not allowed in strict mode.
[Error] Did not parse stylesheet at 'http://allsky.awcloud.ca:8080/bower_components/morrisjs/morris.css' because non CSS MIME types are not allowed in strict mode.
[Error] Did not parse stylesheet at 'http://allsky.awcloud.ca:8080/dist/css/custom.css' because non CSS MIME types are not allowed in strict mode.

theojurriens, 2019-08-20:
same errors in my case

thomasjacquin, 2019-08-20:
Ok that’s probably related to that commit
https://github.com/thomasjacquin/allsky/commit/72719a46038822835b2687716eb4a340453925fe

There’s probably one option to tweak. I’ll take a look.

thomasjacquin, 2019-08-20:
Ok, I pushed an update. Let me know if that fixes it. I changed a couple lines in the web server configuration.

theojurriens, 2019-08-20:
downloaded new conf, rebooted but it doen't help :-(

thomasjacquin, 2019-08-20:
Did you run `sudo gui/install.sh` or did you move it manually to `/etc/lighttpd/lighttpd.conf`?

theojurriens, 2019-08-20:
move it manually

theojurriens, 2019-08-20:
after sudo .... it is working, thanks

pterodyne, 2019-11-04:
I have this issue every time I upgrade.  What is the solution?  I always run sudo ./install.sh in the gui folder.  Should I not be doing that?  Seems like in the past deleting cookies solved the problem but this time around no go.

closed 2021-09-16 by EricClaeys