AllskyTeam/allsky #112: Alternative Camera

across-suffolk, 2019-08-21:
Is it possible to use this software with an alternative camera such as the Altair GP-Cam178M or the QHY5L-II?

thomasjacquin, 2019-08-21:
Hi,

Currently, it is not possible because it's using the ASI library to talk to the camera. However, provided that Altair or QHY have a linux SDK available, it could be integrated. I don't own these cameras so I haven't been able to test it.

universe241981, 2019-08-23:
> Hi,
> 
> Currently, it is not possible because it's using the ASI library to talk to the camera. However, provided that Altair or QHY have a linux SDK available, it could be integrated. I don't own these cameras so I haven't been able to test it.

If you don't mind, I can help to test with QHY camera.
Please tell me how I can make the allsky to control the QHY camera. I have QHY5LIIC and QHY5PIIC for test.

thomasjacquin, 2019-08-23:
There’s a bit of code that needs to be written. Basically, you’d need to make a copy of capture.cpp and change all the ASI specific calls to be QHY specific.
QHY provides a demo cpp file as a reference in their SDK. That’s helpful to know what methods to call.

Then testing can be done. 

If you have experience with C++ development, that shouldn’t be too complex.

across-suffolk, 2020-02-03:
Thanks Thomas. I saw that both QHY and now Altair Astro both have linux SDK available but C++ is beyond me at the moment. I purchased the ASI120mm-s from Astrofest on Saturday and hope to have this running by the coming weekend.

ckuethe, 2020-02-18:
cross-posting to cbassa/stvid#36

Here are a few repos I found last night while looking for python INDI examples. Some of these may be more useful than others for figuring out how to use the [python API](https://github.com/geehalel/pyindi-client)... Not at all looking forward to figuring out how to do this in C++

* https://github.com/teuben/pyASC
* https://github.com/zemogle/pyOculus
* https://github.com/LowellObservatory/All-Sky
* https://github.com/farom57/Orbit-hunter
* https://github.com/oleg-milantiev/allsky.py
* https://github.com/TomMladenov/indi-viewer

closed 2021-09-16 by EricClaeys