AllskyTeam/allsky #113: Damn Dew

dwormuth, 2019-08-25:
Not specifically a software issue, but my prototype is having problems with dew as the night progresses. 

Does anyone have clever suggestions for heaters or other workarounds?

I'm in Syracuse, NY in case the latitude makes a difference.

cbassa, 2019-08-26:
Have a look at [this link](https://www.skyandtelescope.com/astronomy-equipment/diy-dew-heater/) for instructions on how to build a dew heater.

I ended up building a dew heater using six ~27 Ohm resistors powered from 12 V. This generates about 0.9 W of heat and has greatly reduced my problems with dew. I may reduce this to five 27 Ohm resistors for a bit more power, as the dome does still dew over once in a while.

Also have a look at @thomasjacquin's [instructables page](https://www.instructables.com/id/Wireless-All-Sky-Camera/). There's a few cool designs which use temperature and humidity sensors to only turn on the dew heater to keep the inside of the housing a few degrees above the outside temperature.

bbillp, 2019-08-26:
Good info and timely too.  I’ll plan to make such a heater and place it inside the allsky capsule.  Dew builds up around 11PM at my location near the Pacific.     Bill.

ckuethe, 2019-10-20:
[image]

Four 100 ohm, 1/4W resistors in parallel, connected to the raspberry pi 3.3V line. If I need more heat I can put it on the 5V line but it worked nicely last night.

bbillp, 2019-10-20:
I used 10  x 330 ohm 5 watt resistors in parallel with 12volt 2 amp battery for 4 watts of power.

dwormuth, 2019-10-20:
You READ my mind! I was just going to ask if anyone ran any dew heaters off
the Pi. I'm powering the Pi with a "power over ethernet" setup and was
hoping there was a low voltage solution.

I tried a commercial USB power lens heater but it drew too much power and
the Pi got flaky after 2-3 pictures.

BTW, the link mentioned above is:

https://www.skyandtelescope.com/astronomy-equipment/diy-dew-heater/

If someone wants more info. I certainly did.

Dave Wormuth

bbillp, 2019-10-21:
Nope running a heater from the Pi USB is a terrible idea because the heater currant requirements will likely heat up a circuit trace and a micro component on the Pi like a voltage regulator.   Four watts of power is about 800 milliamps at 5 volts which is far more than a USB peripheral.  No wonder the Pi began to die, it was getting ready to SMOKE...... pretty funny .....    

I built two dew heaters from examples on the net, one uses nichrome wire and the other used 10 X 330 ohm resistors which calcs at 33 ohms and when run from 12Volts the power draw is close to 4 watts and around 300 milliamps which is a light load for a 12 volt sealed lead acid burglar alarm battery.   I placed the 10 resistors around the inside perimeter of the Allsky case giving about a 10 degree difference so then I added a small PWM voltage regular to limit the heat.            Bill P.

ckuethe, 2019-10-21:
3.3v through a 25 ohm resistance is only 0.45W, or about 130mA. I didn't need much heat. That was quite enough to keep the dome clear while the rest of everything outside was covered in dew.

But yeah, be mindful of your current draw.

FredoTheGreat, 2019-10-28:
i have fitted one of these https://www.dewcontrol.com/Dew_Heater_Module_-_All_Sky_Camera/p3099125_19045496.aspx

phil-zenger, 2019-10-28:
I was thinking about buying the same heater? How did you connect it to
power?  Did you also add a relay so that you can only turn it in when the
dew reaches a certain level?

Thanks
Phil

FredoTheGreat, 2019-10-28:
Build still in progress but it will be on all the time. It's a very low ampage so used very little power but is plenty warm enough.
12v psu supplying whole unit.

brianboru82, 2020-01-19:
I created an automatic dew heater controlled from the Raspberry Pi using weather data and an internal temperature sensor. I've written it up here, it works pretty well.
[https://www.instructables.com/id/Raspberry-Pi-Dew-Heater-for-All-sky-Camera](https://www.instructables.com/id/Raspberry-Pi-Dew-Heater-for-All-sky-Camera)

bbillp, 2020-01-19:
Very nice, I’m thinking about a motorized sliding lens cover for my telescope and your circuit and Software will get me started.  Bill P.

hdiessner, 2020-02-09:
Based on ckuethe's comment, I created a heater with 6 100ohm resistors and a relay on the 5V pin of the raspberry - controlled via MQTT. Documented here: https://github.com/hdiessner/Allskycam-heating

across-suffolk, 2020-02-11:
I use the DewControl [heater ring](https://www.dewcontrol.com/Dew_Heater_Module_-_All_Sky_Camera/p3099125_19045496.aspx) but Tim Duke at DewControl also sells a small [controller board](https://www.dewcontrol.com/Programmable_Temperature_Controlled_Switch/p3099125_19441359.aspx) that includes a thermometer probe and activates the heater at a programmable threshold. To simplify things I fitted a [Wide Input Shim](https://shop.pimoroni.com/products/wide-input-shim?variant=2168104321034) from Pimoroni and run the whole lot from a single 12v 5amp supply.

pnoah, 2020-02-15:
But you can run the heater through a relay.  Get a relay hat and the heater can be turned off and on without any damage to the Pi.  I run some cooling fans through relays.

DarKOrange75, 2020-08-13:
> Based on ckuethe's comment, I created a heater with 6 100ohm resistors and a relay on the 5V pin of the raspberry - controlled via MQTT. Documented here: https://github.com/hdiessner/Allskycam-heating

Hi, is it possible to put more resistors for more heat? Thank you.

hdiessner, 2020-08-13:
> > Based on ckuethe's comment, I created a heater with 6 100ohm resistors and a relay on the 5V pin of the raspberry - controlled via MQTT. Documented here: https://github.com/hdiessner/Allskycam-heating
> 
> Hi @hdiessner, is it possible to put more resistors for more heat? Thank you.

It will depend on how much current is drawn from the power supply. If the power supply is strong enough and the resistors are distributed so they don't form some kind of fire starting hot spot, more resistors are possible. Just be really careful :)

DarKOrange75, 2020-08-13:
> It will depend on how much current is drawn from the power supply. If the power supply is strong enough and the resistors are distributed so they don't form some kind of fire starting hot spot, more [...]

Thank you, I followed your instructions and powering it from the Rpi which is power by POE. I'll leave it like that it is working well so far, it was just in case. Thank you!

EricClaeys, 2021-09-14:
This issue isn't related to the AllSky software and several suggestions were given, so I'm closing it. Thanks.

closed 2021-09-14 by EricClaeys