AllskyTeam/allsky #114: Timelapse not created - help

BrightskyObservatory, 2019-08-27:
Hi,

I am just getting started with this project using a Raspberry Pi 3B+ and trying to get my head around it all. I was able to capture a nights worth of dew and I couldn't find a timelapse of the night.

I tried running the .scripts/timelapse.sh command in the terminal. It goes through the process of doing something to all the images then says :

./scripts/timelapse.sh: line 25: avconv: command not found
* Deleting sequence
* Timelapse was created
pi@allsky:~/allsky $ 

Any idea what this is about?

Also I tried installing the GUI and it wouldn't work for some reason, not sure if it's me or not. So I am using the text based version, which seems to work just fine.

Great work by the way. First time I have used a Raspberry Pi.

BrightskyObservatory, 2019-08-27:
I cannot remember exactly off hand, sorry. I can always try it again. I know it said something about missing directories or it couldn't make some directories. I am not sure if this was my end though. Sorry I cannot give you any more info, I am still learning. If I get a chance I will try it again. 

By the way, that name change worked in timelapse.sh thank you. It created the timelapse perfectly.

Andrew.

WirthmU, 2019-09-11:
I did got a few timelapses initially but now for several days they do not appear on my website any more.
As suggested above I checked and my timelapse.sh is up to date and using ffmpeg.

If I run timelapse.sh manually after stopping allsky service it seems to work. First it creates all the image files and then starts do make the mpeg file:

......
'/home/pi/allsky/images/20190907/sequence/1721.jpg' -> '/home/pi/allsky/images/20190907/image-20190908062437.jpg'
'/home/pi/allsky/images/20190907/sequence/1722.jpg' -> '/home/pi/allsky/images/20190907/image-20190908062458.jpg'
'/home/pi/allsky/images/20190907/sequence/1723.jpg' -> '/home/pi/allsky/images/20190907/image-20190908062518.jpg'
'/home/pi/allsky/images/20190907/sequence/1724.jpg' -> '/home/pi/allsky/images/20190907/image-20190908062538.jpg'
ffmpeg version 4.1.4-1+rpt1~deb10u1 Copyright (c) 2000-2019 the FFmpeg developers
  built with gcc 8 (Raspbian 8.3.0-6+rpi1)
  configuration: --prefix=/usr --extra-version='1+rpt1~deb10u1' --toolchain=hardened --libdir=/usr/lib/arm-linux-gnueabihf --incdir=/usr/include/arm-linux-gnueabihf --arch=arm --enable-gpl --disable-stripping --enable-avresample --disable-filter=resample --enable-avisynth --enable-gnutls --enable-ladspa --enable-libaom --enable-libass --enable-libbluray --enable-libbs2b --enable-libcaca --enable-libcdio --enable-libcodec2 --enable-libflite --enable-libfontconfig --enable-libfreetype --enable-libfribidi --enable-libgme --enable-libgsm --enable-libjack --enable-libmp3lame --enable-libmysofa --enable-libopenjpeg --enable-libopenmpt --enable-libopus --enable-libpulse --enable-librsvg --enable-librubberband --enable-libshine --enable-libsnappy --enable-libsoxr --enable-libspeex --enable-libssh --enable-libtheora --enable-libtwolame --enable-libvidstab --enable-libvorbis --enable-libvpx --enable-libwavpack --enable-libwebp --enable-libx265 --enable-libxml2 --enable-libxvid --enable-libzmq --enable-libzvbi --enable-lv2 --enable-omx --enable-openal --enable-opengl --enable-sdl2 --enable-omx-rpi --enable-mmal --enable-libdc1394 --enable-libdrm --enable-libiec61883 --enable-chromaprint --enable-frei0r --enable-libx264 --enable-shared
  libavutil      56. 22.100 / 56. 22.100
  libavcodec     58. 35.100 / 58. 35.100
  libavformat    58. 20.100 / 58. 20.100
  libavdevice    58.  5.100 / 58.  5.100
  libavfilter     7. 40.101 /  7. 40.101
  libavresample   4.  0.  0 /  4.  0.  0
  libswscale      5.  3.100 /  5.  3.100
  libswresample   3.  3.100 /  3.  3.100
  libpostproc    55.  3.100 / 55.  3.100
Input #0, image2, from 'images/20190907/sequence/%04d.jpg':
  Duration: 00:01:08.96, start: 0.000000, bitrate: N/A
    Stream #0:0: Video: mjpeg, yuvj420p(pc, bt470bg/unknown/unknown), 1936x1096 [SAR 1:1 DAR 242:137], 25 fps, 25 tbr, 25 tbn, 25 tbc
Stream mapping:
  Stream #0:0 -> #0:0 (mjpeg (native) -> h264 (libx264))
Press [q] to stop, [?] for help
[swscaler @ 0x8efa50] deprecated pixel format used, make sure you did set range correctly
[libx264 @ 0x8650c0] using SAR=1/1
[libx264 @ 0x8650c0] using cpu capabilities: none!
[libx264 @ 0x8650c0] profile High, level 4.2
[libx264 @ 0x8650c0] 264 - core 155 r2917 0a84d98 - H.264/MPEG-4 AVC codec - Copyleft 2003-2018 - http://www.videolan.org/x264.html - options: cabac=1 ref=3 deblock=1:0:0 analyse=0x3:0x113 me=hex subme=7 psy=1 psy_rd=1.00:0.00 mixed_ref=1 me_range=16 chroma_me=1 trellis=1 8x8dct=1 cqm=0 deadzone=21,11 fast_pskip=1 chroma_qp_offset=-2 threads=1 lookahead_threads=1 sliced_threads=0 nr=0 decimate=1 interlaced=0 bluray_compat=0 constrained_intra=0 bframes=3 b_pyramid=2 b_adapt=1 b_bias=0 direct=1 weightb=1 open_gop=0 weightp=2 keyint=250 keyint_min=25 scenecut=40 intra_refresh=0 rc_lookahead=40 rc=abr mbtree=1 bitrate=2000 ratetol=1.0 qcomp=0.60 qpmin=0 qpmax=69 qpstep=4 ip_ratio=1.40 aq=1:1.00
Output #0, mp4, to 'images/20190907/allsky-20190907.mp4':
  Metadata:
    encoder         : Lavf58.20.100
    Stream #0:0: Video: h264 (libx264) (avc1 / 0x31637661), yuv420p, 1936x1096 [SAR 1:1 DAR 242:137], q=-1--1, 2000 kb/s, 25 fps, 12800 tbn, 25 tbc
    Metadata:
      encoder         : Lavc58.35.100 libx264
    Side data:
      cpb: bitrate max/min/avg: 0/0/2000000 buffer size: 0 vbv_delay: -1
frame=  496 fps=0.1 q=21.0 size=    3584kB time=00:00:18.08 bitrate=1623.9kbits/s speed=0.0028x 

However it seems like to take forever. Its been more than two hours since I have started the script and only 18.08 minutes have been processed.

Is it a problem of resources on my RPI B+?

thomasjacquin, 2019-09-11:
I would say 2 hours is not unusual. It all depends on the frames size, the number of images and the Raspberry Pi version.
The time you see is in seconds. 18.08 seconds is the duration of the timelapse. You can also see that it is currently processing the 496th frame (out of X number of images from that night).
However, it seems to be processing 0.1 frame per second and that's pretty low.
On a RPi3 B+ I get more than 1 fps. I haven't tested on the RPi4 yet but I assume it will reach several FPS.

WirthmU, 2019-09-11:
I got a total of 1724 images on that particular night. So I should take another 4 hours.
At least it is continuing for now.

But what could be the reason why the timelapses are not uploaded to my website?
Keograms and startrails are there.

thomasjacquin, 2019-09-11:
The reason could be that UPLOAD_VIDEO is set to false in `config.sh` or that MP4DIR is incorrect in `ftp-settings.sh`.

Also if the timelapse ends with an error it will not upload the video.

WirthmU, 2019-09-11:
The configuration settings should be fine ... I will let run the mpeg creation to go on.
The actual state now is:

frame=  690 fps=0.1 q=21.0 size=    5632kB time=00:00:25.84 bitrate=1785.5kbits/s speed=0.00291x

WirthmU, 2019-09-12:
The file has been created:
frame= 1724 fps=0.1 q=21.0 Lsize=   16311kB time=00:01:08.84 bitrate=1941.0kbits/s speed=0.00307x    
video:16290kB audio:0kB subtitle:0kB other streams:0kB global headers:0kB muxing overhead: 0.129782%
[libx264 @ 0x8650c0] frame I:10    Avg QP:11.71  size: 54143
[libx264 @ 0x8650c0] frame P:446   Avg QP:11.53  size: 16480
[libx264 @ 0x8650c0] frame B:1268  Avg QP:12.94  size:  6931
[libx264 @ 0x8650c0] consecutive B-frames:  1.8%  0.1%  0.9% 97.2%
[libx264 @ 0x8650c0] mb I  I16..4: 31.7% 63.6%  4.7%
[libx264 @ 0x8650c0] mb P  I16..4:  3.6%  9.4%  0.2%  P16..4: 25.0%  5.2%  4.2%  0.0%  0.0%    skip:52.4%
[libx264 @ 0x8650c0] mb B  I16..4:  0.5%  1.7%  0.0%  B16..8: 19.1%  4.0%  0.5%  direct: 2.7%  skip:71.6%  L0:53.6% L1:45.1% BI: 1.2%
[libx264 @ 0x8650c0] final ratefactor: 16.86
[libx264 @ 0x8650c0] 8x8 transform intra:72.4% inter:78.5%
[libx264 @ 0x8650c0] coded y,uvDC,uvAC intra: 38.0% 15.7% 2.9% inter: 6.0% 4.3% 0.2%
[libx264 @ 0x8650c0] i16 v,h,dc,p: 29% 12% 10% 49%
[libx264 @ 0x8650c0] i8 v,h,dc,ddl,ddr,vr,hd,vl,hu: 12%  5% 77%  1%  1%  1%  1%  1%  1%
[libx264 @ 0x8650c0] i4 v,h,dc,ddl,ddr,vr,hd,vl,hu: 27% 18% 40%  3%  4%  2%  2%  2%  1%
[libx264 @ 0x8650c0] i8c dc,h,v,p: 87%  5%  7%  1%
[libx264 @ 0x8650c0] Weighted P-Frames: Y:0.4% UV:0.4%
[libx264 @ 0x8650c0] ref P L0: 52.3%  0.3% 30.3% 17.1%  0.0%
[libx264 @ 0x8650c0] ref B L0: 63.2% 28.0%  8.8%
[libx264 @ 0x8650c0] ref B L1: 83.9% 16.1%
[libx264 @ 0x8650c0] kb/s:1935.07
cd ok, cwd=/web/Saanenmoeser/allsky/videos                                        
16702691 bytes transferred in 3 seconds (5.70 MiB/s)                                   
* Deleting sequence
* Timelapse was created

WirthmU, 2019-09-13:
I did a reboot and let run the Pi over night. Still the same behavior:
Current keogram is uploaded
Current startrail is uploaded
Current timelapse is missing

So if I do a manual run of timelapse.sh the timelapse is created and uploaded to the correct directory - although in a very slow matter.

Somehow timelapse creation as part of the allsky service must be too much of a load for the B+.
I will try to reduce to bin 2.

WirthmU, 2019-09-21:
I am still struggling with timelapse. I did the following:
change to bin 2 > no timelapse during running of allsky service
change delay in settings.json from 10 to 1000 > no timelapse running of allsky service

I checked the allsky.log:

I got plenty of this:

`Sep 21 16:36:14 allsky allsky.sh[17667]: scripts/saveImageDay.sh: line 24: 16962 Aborted                 convert "$IMAGE_TO_USE" -resize 962x720 -gravity East -chop 2x0 "$FILENAME-resize.$EXTENSION"
Sep 21 16:36:55 allsky allsky.sh[17667]: scripts/saveImageDay.sh: line 24: 17033 Aborted                 convert "$IMAGE_TO_USE" -resize 962x720 -gravity East -chop 2x0 "$FILENAME-resize.$EXTENSION"
Sep 21 16:37:31 allsky allsky.sh[17667]: scripts/saveImageDay.sh: line 24: 17088 Aborted                 convert "$IMAGE_TO_USE" -resize 962x720 -gravity East -chop 2x0 "$FILENAME-resize.$EXTENSION"
Sep 21 16:38:13 allsky allsky.sh[17667]: scripts/saveImageDay.sh: line 24: 17151 Aborted                 convert "$IMAGE_TO_USE" -resize 962x720 -gravity East -chop 2x0 "$FILENAME-resize.$EXTENSION"
Sep 21 16:39:00 allsky allsky.sh[17667]: scripts/saveImageDay.sh: line 24: 17224 Aborted                 convert "$IMAGE_TO_USE" -resize 962x720 -gravity East -chop 2x0 "$FILENAME-resize.$EXTENSION"
Sep 21 16:39:36 allsky allsky.sh[17667]: scripts/saveImageDay.sh: line 24: 17308 Aborted                 convert "$IMAGE_TO_USE" -resize 962x720 -gravity East -chop 2x0 "$FILENAME-resize.$EXTENSION"
Sep 21 16:40:12 allsky allsky.sh[17667]: scripts/saveImageDay.sh: line 24: 17363 Aborted                 convert "$IMAGE_TO_USE" -resize 962x720 -gravity East -chop 2x0 "$FILENAME-resize.$EXTENSION"
Sep 21 16:40:59 allsky allsky.sh[17667]: scripts/saveImageDay.sh: line 24: 17436 Aborted                 convert "$IMAGE_TO_USE" -resize 962x720 -gravity East -chop 2x0 "$FILENAME-resize.$EXTENSION"
Sep 21 16:41:41 allsky allsky.sh[17667]: scripts/saveImageDay.sh: line 24: 17507 Aborted                 convert "$IMAGE_TO_USE" -resize 962x720 -gravity East -chop 2x0 "$FILENAME-resize.$EXTENSION"`

and a few one of these:
`Sep 21 16:34:44 allsky allsky.sh[17667]: convert-im6.q16: Premature end of JPEG file `image.jpg' @ warning/jpeg.c/JPEGWarningHandler/387.
Sep 21 16:34:44 allsky allsky.sh[17667]: convert-im6.q16: Corrupt JPEG data: premature end of data segment `image.jpg' @ warning/jpeg.c/JPEGWarningHandler/387.`

I can see from the logfile that mmpeg creation is started but then stops after about a hundred frames

Any ideas?

thomasjacquin, 2019-09-23:
I believe that means it's hitting a bad frame. That could be an image that is corrupted because it was only partially saved. Try deleted the corrupted one and restart the timelapse.

WirthmU, 2019-09-23:
I don't think so. If I do without any deleting ./timelapse.sh by hand the video is created. The mpeg creation only fails when it is part of the morning routine.
Could the timelapse.sh be run as a cron job?

thomasjacquin, 2019-09-23:
The RPi B+ is not as powerful as the new versions so I haven't tested timelapses with this device.
A cron job could work but you'll need to pass a date as an argument to `./timelapse.sh`

thomasjacquin, 2019-09-24:
Yes the RPi 4 works

WirthmU, 2019-10-01:
I did switch to a RPI4 now - installation looked like running smoothly. I get the current picture uploaded to my website regularly.

However, there are no pictures saved on the RPI4. There is not even the picture file created.

Any help to this?

WirthmU, 2019-10-02:
The problem has fixed itself - it just took some time until the images picture folder was created and now everything is working fine.

… ready to deploy outside now!

bortnyk2012, 2021-03-16:
I have this problem too on RPI4, can't to create timelapse local:

pi@allsky:~/allsky/scripts$ ./timelapse.sh 20210315
-----cut------
-enable-libshine --enable-libsnappy --enable-libsoxr --enable-libspeex --enable-libssh --enable-libtheora --enable-libtwolame --enable-libvidstab --enable-libvorbis --enable-libvpx --enable-libwavpack --enable-libwebp --enable-libx265 --enable-libxml2 --enable-libxvid --enable-libzmq --enable-libzvbi --enable-lv2 --enable-omx --enable-openal --enable-opengl --enable-sdl2 --enable-omx-rpi --enable-mmal --enable-neon --enable-rpi --enable-libdc1394 --enable-libdrm --enable-libiec61883 --enable-chromaprint --enable-frei0r --enable-libx264 --libdir=/usr/lib/arm-linux-gnueabihf/neon/vfp --cpu=cortex-a7 --arch=armv6t2 --disable-thumb --enable-shared --disable-doc --disable-programs
  libavutil      56. 22.100 / 56. 22.100
  libavcodec     58. 35.100 / 58. 35.100
  libavformat    58. 20.100 / 58. 20.100
  libavdevice    58.  5.100 / 58.  5.100
  libavfilter     7. 40.101 /  7. 40.101
  libavresample   4.  0.  0 /  4.  0.  0
  libswscale      5.  3.100 /  5.  3.100
  libswresample   3.  3.100 /  3.  3.100
  libpostproc    55.  3.100 / 55.  3.100
Input #0, image2, from 'images/20210315/sequence/%04d.jpg':
  Duration: 00:00:58.88, start: 0.000000, bitrate: N/A
    Stream #0:0: Video: mjpeg, yuvj420p(pc, bt470bg/unknown/unknown), 1280x960 [SAR 1:1 DAR 4:3], 25 fps, 25 tbr, 25 tbn, 25 tbc
[NULL @ 0x126b230] Unable to find a suitable output format for './i'
./i: Invalid argument
./timelapse.sh: line 43: -movflags: command not found
* Deleting sequence
* Timelapse was created
pi@allsky:~/allsky/scripts$ 

any ideas?

thomasjacquin, 2021-03-16:
Can you edit scripts/timelapse.sh and remove the `./i` at the end of the line?
https://github.com/thomasjacquin/allsky/blob/master/scripts/timelapse.sh#L42

This may have been added by accident.
Save the file and run the timelapse manually. Let me know if it helps and I'll remove that typo.

pimmoon, 2021-08-10:
I use 3B+. During the first night I noticed timelapse was not created. I then realized it is because timelapse process crashed my pi when I tried to run the script manaully. So I went ahead and reduce the size of the timelapse to 640x480 and yes. It worked now!! 

So I think the timelapse size can not be too big on older Pi maybe because of less RAM. Anyways, it works now every morning, timelapse is created and it takes a few mins for the process to complete. I notice tho the Pi is running very hot while timelapse is being created. (I see 70c sometimes 80c)

EricClaeys, 2021-09-09:
This issue is from an older release of the AllSky software.  If you think it might still be relevant, please submit a new issue.  Thanks.

closed 2021-09-09 by EricClaeys