AllskyTeam/allsky #115: Graphical Interface Won't Run

jboyles5, 2019-08-31:
Recent setup of the Allsky setup on my Raspberry Pi 4. I keep trying to get the GUI running, but I get connection refused each attempt. I have also tried to load in the local browser on the RPi. 

Attempted to install again and the below is the install log. Any help would be appreciated!
`* Installation of the webserver

Hit:1 http://raspbian.raspberrypi.org/raspbian buster InRelease
Hit:2 http://archive.raspberrypi.org/debian buster InRelease
Hit:3 http://linux.teamviewer.com/deb stable InRelease
Reading package lists... Done
Reading package lists... Done
Building dependency tree
Reading state information... Done
avahi-daemon is already the newest version (0.7-4+b1).
hostapd is already the newest version (2:2.7+git20190128+0c1e29f-6).
lighttpd is already the newest version (1.4.53-4).
php7.3-cgi is already the newest version (7.3.4-2).
dnsmasq is already the newest version (2.80-1+rpt1).
0 upgraded, 0 newly installed, 0 to remove and 12 not upgraded.
already enabled
Run "service lighttpd force-reload" to enable changes
Job for lighttpd.service failed because the control process exited with error code.
See "systemctl status lighttpd.service" and "journalctl -xe" for details.

* Configuring lighttpd

* Changing hostname to allsky

* Setting avahi-daemon configuration

* Adding the right permissions to the web server

* Retrieving github files to build admin portal
Cloning into '/var/www/html'...
remote: Enumerating objects: 4, done.
remote: Counting objects: 100% (4/4), done.
remote: Compressing objects: 100% (4/4), done.
remote: Total 1771 (delta 0), reused 1 (delta 0), pack-reused 1767
Receiving objects: 100% (1771/1771), 2.72 MiB | 5.96 MiB/s, done.
Resolving deltas: 100% (610/610), done.
mkdir: cannot create directory '/etc/raspap': File exists

* Modify config.sh

The Allsky Portal is now installed
You can now reboot the Raspberry Pi and connect to it from your laptop, computer, phone, tablet at this address: http://allsky.local
`

thomasjacquin, 2019-08-31:
I will give it a try tonight on the RPi4 I just received. It seems that lighttpd.conf may have an error in it and the service fails to start properly.

jboyles5, 2019-08-31:
@thomasjacquin thanks. let me know if you need me to try anything.

thomasjacquin, 2019-08-31:
Alright, there was a syntax error. I have fixed the code. Can you try re-installing it?

jboyles5, 2019-08-31:
@thomasjacquin do I need to remove anything or just run gui/install.sh again?

thomasjacquin, 2019-08-31:
The following should be enough:
`git clone https://github.com/thomasjacquin/allsky.git`
`cd allsky`
`sudo gui/install.sh`

jboyles5, 2019-08-31:
@thomasjacquin i had to do a clean install of allsky, but it worked after that. thanks!

closed 2019-08-31 by jboyles5