AllskyTeam/allsky #116: Dark frame issue

Fulhair, 2019-09-01:
I've got my camera up and running but finding the video isn't great and I'm getting a lot of artifact on the video as colored pixels. I've tried multiple attempts at updating the dark frames as per the instructions but that hasn't improved it. 

Here's an example video showing the problem:  https://drive.google.com/open?id=1v3hjIxtA4BpY6-H3a3VyGDvz_0V4zC3n

Any suggestions how to improve it?
Cheers

Also the public.php picture link is broken

thomasjacquin, 2019-09-01:
Darks are temperature and exposure/gain dependant. Currently you can only chose one dark but I will add a temperature dependant dark library that gets applied based on current settings.

Since you have auto-exposure enabled and most your images are at 20s, you should take a 20s dark at around the same outside temperature you get overnight.
It won’t look perfect at the beginning and end of night but it should get rid of the subtraction artifacts.

thomasjacquin, 2019-09-05:
The public.php page should be fixed now.

EricClaeys, 2021-09-09:
This issue is from an older release of the AllSky software.  If you think it might still be relevant, please submit a new issue.  Thanks.
Darks were improved in version 0.8 to take pictures with the specified night settings.

closed 2021-09-09 by EricClaeys