AllskyTeam/allsky #119: added iris close for dark frame generation

kpalivec, 2019-09-19:
I'm new to github so if I'm posting this in the wrong place or not following proper procedure please forgive me! 

I had a lens with an auto iris cable and jumped it to ground and gpio pin 4. Installed wiringpi and edited allsky.sh to make gpio4 go low when a dark frame generation is requested. normally it stays high to keep the iris full open on the lens. 
---------------------------------------------------
edited allsky.sh with dark frame iris control added 
----------------------------------------------------
`
#!/bin/bash
source /home/pi/allsky/config.sh
source /home/pi/allsky/scripts/filename.sh
#set gpio pins for output
gpio mode 7 out

checkDark="darkframe 1"

echo "Starting allsky camera..."
cd /home/pi/allsky

# Building the arguments to pass to the capture binary
ARGUMENTS=""
KEYS=( $(jq -r 'keys[]' $CAMERA_SETTINGS) )
for KEY in ${KEYS[@]}
do
        ARGUMENTS="$ARGUMENTS -$KEY `jq -r '.'$KEY $CAMERA_SETTINGS` "
done

# When using a desktop environment (Remote Desktop, VNC, HDMI output, etc), a preview of the capture can be displayed in a separate window
# The preview mode does not work if allsky.sh is started as a service or if the debian distribution has no desktop environment.
if [[ $1 == "preview" ]] ; then
        ARGUMENTS="$ARGUMENTS -preview 1"
fi
ARGUMENTS="$ARGUMENTS -daytime $DAYTIME"

echo "$ARGUMENTS">>log.txt
# Check to see if we're collecting a DarkFrame and if so. Close the iris.
# If not. Open it

if echo $ARGUMENTS | grep -q "$checkDark";

then
   echo "Creating DarkFrame: Iris Closed"
   gpio write 7 0
else
   echo "Normal Operation: Iris Open"
   gpio write 7 1
fi
./capture $ARGUMENTS

`

thomasjacquin, 2019-09-19:
Thanks for that piece of code. The auto-iris is definitely an improvement over the manual method. Especially if the camera is already on the roof.
What lens model are you using in your allsky?

kpalivec, 2019-09-20:
The Lens is a Rainbow L163VDC4P fisheye lens. It had a connector for an auto iris connection to a camera. I broke out the pins and isolated just the iris shutter coil power leads. (the iris normally works on some kind of push/pull feedback system with viltages provided by the cctv camera based on video levels. since all I wanted was to be able to close it all the way or open it all the way. all I was concerned with was the actual iris so gpio level voltages on the + - leads were enough to accomplish what I needed.

EricClaeys, 2021-09-09:
This issue is from an older release of the AllSky software.  If you think it might still be relevant, please submit a new issue.  Thanks.

closed 2021-09-09 by EricClaeys