AllskyTeam/allsky #120: better detection and rejection of corrupt images

ckuethe, 2019-09-24:
TL;DR - need a way to detect and delete images with corrupt jpeg data to prevent them from messing up startrails and keograms.

For reasons not clear to me, some of the jpgs written to disk are corrupt, causing startrails to complain like this:

```
[5367/8285] image-20190924030959.jpg 0.142903
[5368/8285] image-20190924031003.jpg 0.142869
Corrupt JPEG data: bad Huffman code
[5369/8285] image-20190924031007.jpg 0.425546
[5370/8285] image-20190924031011.jpg 0.142781
[5371/8285] image-20190924031015.jpg 0.142778
```

and:

```
[6918/8285] image-20190924045609.jpg 0.150468
[6919/8285] image-20190924045613.jpg 0.15043
Corrupt JPEG data: 1146 extraneous bytes before marker 0xd9
[6920/8285] image-20190924045617.jpg 0.204732
[6921/8285] image-20190924045621.jpg 0.150463
[6922/8285] image-20190924045625.jpg 0.150506
```

which leads to strange startrail images like this.

[image]

The only possibly weird thing in dmesg is a few of these:

```
[134352.478664] usb 1-1.3: reset high-speed USB device number 6 using dwc_otg
[138920.669298] usb 1-1.3: reset high-speed USB device number 6 using dwc_otg
[141612.344527] usb 1-1.3: reset high-speed USB device number 6 using dwc_otg
```

but nothing in the allsky capture log indicating a capture error or failure to compress.

If I delete all the images with corrupt jpeg data and rerun `startrail` I get a much better looking image.

[image]

thomasjacquin, 2019-09-24:
Regarding the message `reset high-speed USB device number 6 using dwc_otg`, this is basically saying that the connection to the camera was reset because of some issue (low voltage could be causing it).
I had this kind of issue happen to me on a RPi4 using the USB3 and powered with the RPi3 power supply (2.5V instead of 3V).
This could also happen because of a long or low quality USB cable.

ckuethe, 2019-09-24:
I'm using the USB cable that came with my ZWO ASI120MM-S, but I'll see about swapping it for a shorter cable.

Was just checking into the power supply; I made sure to get a good 3A, 5.2V supply (to account for cable loss), and neither `dmesg` nor `vcgencmd get_throttled` give any indication of undervoltage events.

dbainbridge, 2019-09-24:
FYI, for long power runs I've used https://www.amazon.com/gp/product/B018X0BAYS.. so I'm actually sending 48V about 500ft and using that to convert it to 5V.

ckuethe, 2019-09-24:
My power run is only 5ft. But thanks for the pointer to that buck converter, it might be useful for some other outdoor projects I'm working on.

closed 2019-10-04 by thomasjacquin