AllskyTeam/allsky #127: Delay between images

kalalaa, 2019-10-04:
Hi!

I made the allsky camera with T7c and 3B+. It worked fine for the first time. No need to install asi firmware (factory installed asi120), and no need to add line into config file. 
But i have a big problem. I can't set delay between images. I tried everything:
-with or without auto exposure
-set delay
-set auto exposure
Always take the image without any delay. The night delay can't modify too.
Tried write manuially the settings.json file too, but nothing works.
Any suggestion?

Thanks!

thomasjacquin, 2019-10-04:
Well this is funny because the delay is one of the camera settings that has nothing to do with the firmware. It is just a value in milliseconds that gets used between each shot.
You can see where it's used in `capture.cpp`. The variable is called `useDelay `.
What values are you using for day  and night delays?

kalalaa, 2019-10-05:
I want 60 second for night and anything for day. The stock value for day is fine. I found if i set anything in the graphical interface and save, it woll not set in the settings.json file. 
Maybe i can set it in the capture.cpp.

kalalaa, 2019-10-05:
A new info: i have graphical interface installed, but i haven't the settings.json file in /var/www/html. Maybe i haven't permission to modify this folder? Try to reinstall everything? Or can i rewrite somewhere the delay time? (capture.cpp)

thomasjacquin, 2019-10-06:
I just noticed the README file is not completely up to date. The file `settings.json` is now located at `/etc/raspap/settings.json` after you install the GUI.
When you save your settings using the GUI, this is the file that should be modified.

kalalaa, 2019-10-06:
settings.json located at /etc/raspapp/ , and it save the GUI setting. So it is ok.
In capture.cpp i modified useDelay = 0 to = delay
Now if i modify daytime delay it set day and night delay. I don't care the daytime delay, so now i can use it how i want.
I don't know it works like this because i modified the usedelay or it works this way before that. But now i don't care. It works.
Thank you for the development. It's a really good stuff. Now i can see the clouds during astrofotographie.
Thank you!

kalalaa, 2019-10-06:
Oh sorry! I wrote it too soon. I changed the daytime delay to 100 in gui. It worked. Changed back to 60000, but nothing. It shoots again as long as the exposure set. No delay. So not the usedelay changing worked. i changed it back to = 0 (tried with 60000 too but nothing) 
But i don't know anything about programming, so changed everything back to original.

I don't know how can i set 60s delay before, but i can't repeat it.
If you have any tips let me know.
I want to use 30s expo without auto exposure, and 60s delay. 

My only problem is the too many pictures at one night, so if you don't have any tips it's ok.  I will set the auto deleting to less days.

kalalaa, 2019-10-07:
Everything is ok! Delay works. After it switches from day to night it uses delay. So i don't touch anything. :-)

EricClaeys, 2021-09-09:
This issue is from an older release of the AllSky software.  If you think it might still be relevant, please submit a new issue.  Thanks.
It sounds like you resolved the issue.

closed 2021-09-09 by EricClaeys