AllskyTeam/allsky #131: FTP to NAS drive

FredoTheGreat, 2019-10-21:
I have tried to set the FTP to my network drive (ReadyNAS) but I am getting a certificate untrusted error.

I have tried adding set ssl:verify-certificate false to one of the scripts as a test but it says its an unknown command!
any assistance gratefully received

FredoTheGreat, 2019-10-22:
this is the error

cd: Fatal error: Certificate verification: Not trusted (2A:F1:93:C7:A2:7C:C5:6A:1A:03:D9:9C:48:F2:1D:75:8E:F0:48:38)
put: images/20191022/allsky-20191022.mp4: Fatal error: Certificate verification: Not trusted (2A:F1:93:C7:A2:7C:C5:6A:1A:03:D9:9C:48:F2:1D:75:8E:F0:48:38)

thomasjacquin, 2019-10-22:
The option should work if you add it there for example:
`lftp "$PROTOCOL"://"$USER":"$PASSWORD"@"$HOST":"$MP4DIR" -e "ssl:verify-certificate false; set net:max-retries 1; put $TIMELAPSE; bye"`

Have you tried to SSH to the destination directly? `ssh <user>@<NAS_IP_address>`. It may ask you if you want to add this host as a trusted host.

FredoTheGreat, 2019-10-22:
Ok thanks I will give it a go

Dave Wilkes

FredoTheGreat, 2019-10-23:
I get this with that line in!
pi@allsky:~/allsky/scripts $ cd: Fatal error: Certificate verification: Not trusted (2A:F1:93:C7:A2:7C:C5:6A:1A:03:D9:9C:48:F2:1D:75:8E:F0:48:38)
Unknown command `ssl:verify-certificate'.

if i try ssh to the nas i get connection refused

i think there is an issue with the nas tbh when i connect with a web browser it shows an ssl error and the ip address in the certificate is nothing like what it should be. also reading up on these nas drives there are lot of people with certificate issues. i will look at another option. i see someone has used drop box!

FredoTheGreat, 2019-10-28:
I gave up on the ReadyNAS they are not updating it anymore so will not fix their poor SSL setup.
i have now configured it to SFTP to my web hosted SFTP server and all works well.

dl8ndg, 2019-11-28:
I am facing the same issue with fritz.nas (NAS at Fritz.Box from AVM).
Since it is not possible to accept the certificate, FTP-uploads are not possible.
At the same time internal recording stops.
It seems there is no way to correct the problem, by disabling uploads or setting ftp-settings.sh to default.

thomasjacquin, 2019-12-01:
> It seems there is no way to correct the problem, by disabling uploads or setting ftp-settings.sh to default.

You can disable upload by setting `UPLOAD_IMG` to false in config.sh

dl8ndg, 2019-12-07:
Because it is ftp and not sftp, I changed the settings as describes a lot of threads ago:

I replaced
lftp "$PROTOCOL"://"$USER":"$PASSWORD"@"$HOST":"$IMGDIR" -e "set net:max-retries 1; set net:timeout 20; put $FILENAME-resize.$EXTENSION; bye" &
by
lftp "$PROTOCOL"://"$USER":"$PASSWORD"@"$HOST":"$IMGDIR" -e "set ssl:verify-certificate false; set net:max-retries 1; set net:timeout 20; put $FILENAME-resize.$EXTENSION; bye" &
in scripts/saveImageDay.sh, scripts/saveImageNight.sh and scripts/timelapse.sh

now ftp works, but the image is permanently written into the root folder of the ftp-server.
settings like IMGDIR='allsky' are ignored.

dnuber, 2019-12-30:
I had same problem - would only save to root of host (external server, not internal NAS) when set ssl:verify-certificate false. The sftp/certificate thing, especially when dealing with self-signed is a PITA. But bypassing is not a good solution (ie. opens you up to man-in-middle attacks).

Managed to resolve by doing this (and not using ssl:verify-certificate false):

Create RC file (put in /home/pi/.lftp/rc ):
#######################
# RC file
set ssl:ca-file "/home/pi/.lftp/cert_rasc_webnames.crt"
#
set ssl:check-hostname no
#########################

Create cert file in /home/pi/.lftp/ called cert_rasc_webnames.crt (and make own/grp "pi"
Get content for Cert file from:
openssl s_client -showcerts -connect example.com:21 -starttls ftp
replace example.com with the host name of your ftp server and change the port number if different.

This will return a bunch of information, the bit we are interested in is under Certificate chain section. Copy the certificate which is the text which starts -----BEGIN CERTIFICATE----- and ends -----END CERTIFICATE----- and paste into a new file under  ~/.lftp eg. mycert.crt so you now have a file that reads:

-----BEGIN CERTIFICATE-----
<lots of characters...>
-----END CERTIFICATE-----

Hope this helps. I did a manual sftp command from terminal window, before and after, to verify it was working.

DaveNL

EricClaeys, 2021-09-14:
This issue is from an older release of the AllSky software. If you think it might still be relevant, please submit a new issue.
Also, this appears to be related to certain NAS devices, not to the AllSky software.

closed 2021-09-14 by EricClaeys

mpinnuck, 2022-02-18:
To file upload.sh inserted "echo set ssl:verify-certificate false" after line 120 fixed the problem

EricClaeys, 2022-02-18:
@mpinnuck You shouldn't change the `upload.sh` file.  The Wiki page on Troubleshooting uploads describes where you can put the "echo set ssl:verify-certificate false" line where it won't get overwritten every time you update Allsky.

mpinnuck, 2022-02-20:
Thanks, added to LFTP_COMMANDS in ftp-settings.sh

cr0mulent1, 2023-12-15:
"Have you tried to SSH to the destination directly? ssh <user>@<NAS_IP_address>. It may ask you if you want to add this host as a trusted host."

This worked for me!!