AllskyTeam/allsky #132: How do I save generated keograms and startrails to different directory

wannaberocker, 2019-10-27:
Hello, I have figured out how to make timelapses save to the directory I choose but I cannot figure out which lines to edit to have the star trails and keograms save to the directory of my choosing.

What lines of code do I need to edit to make this happen and where are they located? Thanks.

ckuethe, 2020-08-01:
Look for the calls to `keogram` and `startrails` in [`endOfNight.sh`](https://github.com/thomasjacquin/allsky/blob/master/scripts/endOfNight.sh)

EricClaeys, 2021-09-05:
This issue is from an older release of the AllSky software and appears to have been resolved or a possible resolution given.  If you think it might still be relevant, please test it with the newest version and submit a new issue if needed.  Thanks.

closed 2021-09-05 by EricClaeys