AllskyTeam/allsky #133: High temperatures

ZebrasHaveBigEars, 2019-11-08:
Hi Everyone

I finally got a redesigned allsky system on my roof and it is working really well. However the temperature of the camera is now recording @ 70 degrees centigrade during hot days and it is still Spring over here in Australia. Does anyone see a problem with running at these kind of temperatures or even higher as we head into Summer with occasional days at another 10 + degrees outside.

I guess I could add some fans but this would reduce the weatherproofing. Any ideas?

Cheers

ZHBE

rogergau, 2019-11-08:
I have had exactly the same thought. In Western Australia I still have my allsky cam mounted on the inside of the observstory so it’s protected during the day from extreme heat. This is not ideal of course, I would like to see the sky no matter the time of day or weather conditions.

I am in the process of making a housing but haven’t fully decided on the way to extract heat from it. I have wondered about shaded passive heat sink (metal aluminium of sorts perhaps with actual heat sink fins attached) on the bottom of the housing, metal contact to the camera body, and the raspberry pi safely inside, not in the housing outside.

I’d be interested if you come up with a neat solution.

Roger.

RobertG73, 2019-11-08:
Hi, i had the same problem 65+ degrees Celsius inside the housing. I am using now a Peltier unit 40x40mm at 12V with the current limited to 0,6Amps using a 20 Ohm 20W ceramic resistor. Temperature dropped to 50 degrees but i guess to the next summer i will need to raise to 1Amp. Now, the winter is coming here, and i stopped the experiments. I am using a 12V 5A power supply so i have where from to raise the current to Peltier in the future.

ckuethe, 2019-11-23:
[image]

My skycam is cooled with with a 40mm fan. The intake is on the bottom, and it exhausts out the side. Both vent holes have a fine mesh over them to keep out insects. I might add some louvers on the side and a pipe on the bottom to make it very difficult to draw in water from the bottom, and  to keep motivated raindrops from sneaking in the the out door...

RobertG73, 2019-11-24:
I put a picture with my build, where it can be seen the 40mm fan which blow the air on the cold surface of the Peltier element.
[image]

pnoah, 2020-02-15:
I also had overheating problems in the summer.  I added relays and two 12V fans.  I added some code to turn the fans on and off based on the temperature.  Worked well and kept the temperature way down.  I mounted the fans on the bottom of the case so there is no water entering.  I also added a display to the image to indicate which fan is on or off as well as case temperature and camera temperature.

> Hi Everyone
>
> I finally got a redesigned allsky system on my roof and it is working really well. However the temperature of the camera is now recording @ 70 degrees centigrade during hot days and it [...]

EricClaeys, 2021-09-05:
This is not related to the AllSky software and possible resolutions have been proposed.  If you think it might still be relevant, please test it with the newest version and submit a new issue if needed.  Thanks.
FYI, In version 0.8 of AllSky, there's an option to use a different exposure method that can lower the camera temperature about 10 degrees.  In the GUI, go to the bottom and click on "Show advanced options", then check the "Version 0.8 Exposure" box.  Please note that when some people use this option, they get ASI_ERROR_TIMEOUTs in /var/log/allsky.log and are unable to capture any images.  In that case, simply UNcheck the box.

closed 2021-09-05 by EricClaeys