AllskyTeam/allsky #135: Installation guide

michelprog, 2019-11-21:
Hello,

I am trying to get allsky running an a raspberry 3, but i have some problems getting it to compile, for example:

capture.cpp:1:33: fatal error: opencv2/core/core.hpp: No such file or directory

I compiled and installed opencv 4.1.1 from the sources, and tried to include opencv headers, but the problem remains. Is there a recommend way to get allsky to compile? Should i install a specific version op opencv?

thomasjacquin, 2019-11-21:
Hi Michel,

You shouldn't have to do any special OpenCV installation. Are you using the latest Raspbian (Buster)?
Are there any other errors when running the install script?

michelprog, 2019-11-22:
It is solved now. I think i ran the install script before upgrading to buster (the pi was running jessie), and assumed i had to manually install openCv.

 After rerunning the install script, allsky installs and works without problems.

Thanks for your quick response,
Michel

closed 2019-11-22 by michelprog