AllskyTeam/allsky #136: Real Time Clock without the Internet

bbillp, 2019-11-26:
The question has probably been asked before but the issue came up when I took my AllSky out to a Astro Club Star Party far from the internet, no date/time so no images.  Is there a fool proof way to force imaging without current date / time ?   I simply want to apply power and let it image for the night.    

Thanks .... Bill in Orange County

ckuethe, 2019-11-26:
You can get a real time clock chip for the raspberry pi. It plugs right on to the expansion header and then you have an offline clock like a "real computer"

bbillp, 2019-11-27:
I Am aware of various RTC board solutions but would rather change a parameter in the settings file for this setup.    Thanks ...

thomasjacquin, 2019-11-27:
I assume that you could use your phone as a hotspot. If it has previously been used, the RPi should connect to it and get the time from the internet. It should keep the proper time until you unplug it.

I haven't tried it so that's just a guess.

ckuethe, 2019-11-27:
The pi doesn't have a real RTC. There is `fake-hwclock` which writes a timestamp to disk every so often which will at least make the pi have a time ... in this millenium. Unfortunately that means that if you power off the pi at noon, set up in the field at 6pm to start imaging at 7... the pi will basically think it's 1pm when it should be 7.

Try setting some very high elevation in your settings file, eg. 89.9 degrees, so that suncalc will always think the sun is below your arbitrary horizon - that might do what you want.

bbillp, 2019-11-27:
The iphone hotspot and Pi have been used in various Apps but the Star Party Location has zero cell phone signals so I’m not sure if Time of Day will come off the iphone/ipad.  I don’t need time of day for the night out, I just want the imagine but I will try a GPS hookup because I already have a Garmin with USB and if that doesn’t work then I’ll try a RTC board.  UUGEAR WittyPi Power and RTC board also works great but it must be setup at home before leaving to the remote site and it requires a taller Case for the Pi.    

Oh8stn.net has detailed instructions for RTC and/or USB GPS clock input.

I’ll post results.      Thanks ...... Bill

ckuethe, 2019-11-27:
A gps will work very well, I recommend using `chrony` as your time server daemon rather than `ntpd`. I'm using a gps+chrony on my skycam, no NTP required.

bbillp, 2019-11-28:
Oh8stn has an excellent youtube video for the rtc and gps. I got the rtc installed aand working this afternoon and will order a USB gps and will take the pi out of WiFi range to test again late tonight. There is a question about how long the super capacitor will supply standby power too. The rtc used in the video was 2.50 USD and was in my junk box by luck.  Unfortunately the Garmin gps60 will not output NMEA position data from its USB port, only from its serial port so signal conversion must be added and the Pi host serial port definition has to be changed for 9600 serial input. All doable butnot today.

bbillp, 2019-11-28:
Thanks, I missed seeing this message and it helps explain that parameter but now I have the RTC working.    Bill.

bbillp, 2019-11-28:
This is helpful, the UUGEAR WittyPi Mini power management and RTC board uses a similar SuperCapacitor for time clock backup and notes the backup time is about 17 hours.

bbillp, 2019-12-01:
Setting the elevation angle to 89.9 worked fine, thanks.

rave3c0, 2019-12-10:
> I assume that you could use your phone as a hotspot. If it has previously been used, the RPi should connect to it and get the time from the internet. It should keep the proper time until you unplug it.
> 
> I haven't tried it so that's just a guess.

This will work.  I occasionally have to do it because of home wifi issues.  Also have used it while away from home too.

What works for me is setting my phone hotspot name to a known wifi with the same password and encryption such as my home network and it connects right up.  Then you can use a terminal app on your phone, locally, and run `ip neigh` which will give you any IPs and MACs connected to the hotspot.  You'll get your allsky IP and be able to use the GUI webpage too.  Also, you do not have to run the hotspot all night either, once the pi gets the time its good to go!  Just make sure the new image date folder is created first before terminating the hotspot.

EricClaeys, 2021-09-05:
This issue is from an older release of the AllSky software and the issue appear to have been resolved, or possibly resolutions suggested.  If you think it might still be relevant, please test it with the newest version and submit a new issue if needed.  Thanks.

closed 2021-09-05 by EricClaeys