AllskyTeam/allsky #139: No internal storage of pictures after ftp initialisation

dl8ndg, 2019-11-28:
After initialising ftp by setting ftp-settings.sh and enabling upload in config.sh
no more pictures are store internally on the raspi.
Because ftp fails due to the lack of unaccepted certificates,
I tried to go back to internal storage, by disabling uploads in config.sh and setting ftp-settings.sh to the default values.
But, this fails. Pictures are not stored internally any longer.

thomasjacquin, 2019-12-01:
Uploading and saving images should not affect each other. If UPLOAD is set to false in config.sh, the images should save to disk. Have you seen any errors in `service allsky status` ?

dl8ndg, 2019-12-07:
The problem occured, whe the ftp upload to the server failed.
(ftp-problem due to missing certificate)
may be a consecutive fault?

EricClaeys, 2021-09-14:
This issue is from an older release of the AllSky software. If you think it might still be relevant, please submit a new issue. Thanks.

closed 2021-09-14 by EricClaeys