AllskyTeam/allsky #140: Saving of pictures stops after deleting image folder

dl8ndg, 2019-11-29:
After I deleted an old image folder in allsky->images with an FTP client,
the saving of new images into the image folder stoped. 
Live pictures are still available in the allsky folder.

I think, I observed this problem also, after deleting an image folder with the GUI  in the images menu.

Raspi reboot or restarting the service are not resolving the problem.

HW Raspi 3B
OS Buster Lite

thomasjacquin, 2019-12-01:
What's the output of `service allsky status`? Anything in the log file?

dl8ndg, 2019-12-02:
Hi Thomas,
the allsky service was running.
Since I changed to the RasPi 4B meanwhile, I can't evaluate the problem with the RasPi 3B any longer.
I will check with the 4B what will happen after deleting a folder in the next days.
Regards
Axel

EricClaeys, 2021-09-14:
This issue is from an older release of the AllSky software. If you think it might still be relevant, please submit a new issue. Thanks.

closed 2021-09-14 by EricClaeys