AllskyTeam/allsky #141: Timelapse-file is corrupted

dl8ndg, 2019-11-30:
Timelapse *.mp4 file is created, but the filesize is only 48bytes.
of course, it is not possible to play this file,

bbillp, 2019-11-30:
Check the file name Font case, it didn’t match the script Font case, no output.  I had that problem last year.

dl8ndg, 2019-12-01:
No, it has nothing to do with the font case.
It is depending on the used hardware.
I used the same Raspbian Buster Lite image for 3 installations.
The corrupted 48byte timelapse file was created on  RasPi 2B and RasPi 3B.
On the RasPi 4B, the timelapse file was created without any problem.

thomasjacquin, 2019-12-01:
What happens when you run the timelapse manually? What camera are you using?
Some cameras with large sensors output images that are too large for older RPis to cope with during the ffmpeg process (out of memory issue).

dl8ndg, 2019-12-02:
Hi Thomas,
I'm using the ZWO ASI 178MC with 3096x2080Pixel in color mode.
I think you are absolutely right, it's an out of memory problem, since with the 4GB RasPi 4B everything runs fine now.
Best Regards,
Axel

dl8ndg, 2019-12-07:
Meanwhile I've got also a corrupted timelapse file at the RasPi 4.
Most probably, the rootcause for the corrupted file is the attempt of a read access to the timelapse file during the creation process.
That means, if you try to open the timelapse file, when the timelapse creation process is still running, it seems the process is aborted and the created file is unreadable.
Restarting the timelapse process manually with the same picture set - and patience (opening the MP4 after the process finished) created a readable file.

dl8ndg, 2019-12-07:
Additional remark: 
on the RasPi2 and 3 the timelapse process takes much longer - so, there is a good chance, if you open the MP4 in the morning short after sunset, that the problem described above occurs.

ossvale, 2019-12-11:
I get always a segmentation fault problem at the end of the timelapse mp4 creation. To tell the truth it doesn't always happens... sometimes the mp4 gets created and sometimes it is shorter and unreadable.
Trying to re-launch the timelapse script again and again doesn't solve the problem, just a different size mp4 file but still unreadable.
enclosed the output of the timelaps script
[image]

ossvale, 2019-12-16:
I found the the error is due to memory so I increased the swap file and now it seems to work fine

EricClaeys, 2021-09-14:
This issue is from an older release of the AllSky software. If you think it might still be relevant, please submit a new issue. Thanks.
FYI, the README.md file mentions a couple things to do if you are having problems creating timelapse files.

closed 2021-09-14 by EricClaeys