AllskyTeam/allsky #142: SFTP

Henni-S, 2019-12-02:
Hello, I try to set a SFTP connection but i'ts very tricky.
SFTP has not the standard port:21 it has port 22. So I add the 
./scripts/ftp-settings.sh

`
PROTOCOL='sftp'
PORT='22'
USER='user'
PASSWORD='pw'
HOST='ssh-host'
IMGDIR='/allsky/'
MP4DIR='/allsky/videos/'
KEOGRAM_DIR='/allsky/keograms/'
STARTRAILS_DIR='/allsky/startrails/'
`
than I add the lftp line in ./scripts/saveImageNight.sh -p"$PORT"

`lftp -p "$PORT" "$PROTOCOL"://"$USER":"$PASSWORD"@"$HOST":"$IMGDIR" -e "set net:max-retries 1; set net:timeout 20; put $FILENAME-resize.$EXTENSION; bye"`

Now the SFTP connetion works, but I get the next error:
cd: Zugriff nicht möglich:No such file (/allsky)

SFTP works with my Pi:
$ lftp sftp://[user]@[host]:22
passord:
now I can play on the host.

So long!

thomasjacquin, 2019-12-02:
Do you have an `allsky` directory on your host? You will have to create it if it doesn't exist otherwise the sftp upload will return an error.

Henni-S, 2019-12-03:
Yes the setting works with ftp. I can see the lifeview at my website. Only sftp doesn't work.

FredoTheGreat, 2019-12-04:
I can use sftp but have only changed the setting in ftp-settings.sh to sftp 
# FTP/SFTP settings
PROTOCOL='sftp'

Henni-S, 2019-12-05:
It works now. The problem was the unconfirmed SSH certificate key. Hereby I could solve the problem:
[https://www.wizlab.it/code/lftp-fix-fatal-error-host-key-verification-failed.html](url)
But the syslog still shows the error on every uploaded image:

$ tail /var/log/syslog
Dec  5 20:34:50 allsky allsky.sh[1157]: cd: Access failed: No such file (/allsky)

What could that be?

So long, Henri

thomasjacquin, 2019-12-06:
Henri, can you post the content of `ftp-settings.sh` (without the sensitive information). I just want to see the structure of your PATH on your host.

Henni-S, 2019-12-07:
Hi Thomas.
I think I have it now. I have to put a /www/ in front of the directories in ./scripts/ftp-settings.sh.
Now there are no more errors in the syslog.
The last 4 lines from ftp-settings.sh:

ImgDir='/www/all-sky/'
MP4DIR='/www/all-sky/videos/'
KEOGRAM_DIR='/ www/all-sky/keograms/'
STARTRAILS_DIR='/www/all- sky/startrails/'

Best regards, Henri

nielsiwilsi, 2019-12-07:
Hi,
There is a space in your script, use this line.

KEOGRAM_DIR='/www/all-sky/keograms/'

Niels

Henni-S, 2019-12-07:
Oh thanks, Niels, you have a sharp look! 
Best regards, Henri

laheller, 2020-04-08:
Hi! @thomasjacquin @Henni-S 

I have similar problem, but my upload does not work for any protocol. Tried both ftp/sftp, ports 21/22. Also tried the trick with ssh.

Error message is:

> Starting daytime capture
> Saving auto exposed images every 5000 ms
> 
> Press Ctrl+C to stop
> 
> Exposure value: 64 µs
> Saving...20200408 22:41:24
> Auto Exposure value: 0 ms
> Resizing
> Uploading
> 
> cd: Fatal error: Certificate verification: Not trusted (04:99:53:71:10:DC:A8:FD:FB:30:66:8E:69:DF:6A:FD:25:61:1F:6F)
> put: image-resize.jpg: Fatal error: Certificate verification: Not trusted (04:99:53:71:10:DC:A8:FD:FB:30:66:8E:69:DF:6A:FD:25:61:1F:6F)
> 

This is my ftp config file:

> #!/bin/bash
> 
> # FTP/SFTP settings
> PROTOCOL='ftp'
> PORT='21'
> USER='UserName'
> PASSWORD='*****'
> HOST='rigel.sk'
> IMGDIR='/sub/allsky/'
> MP4DIR='/sub/allsky/videos/'
> KEOGRAM_DIR='/sub/allsky/keograms/'
> STARTRAILS_DIR='/sub/allsky/startrails/'
> 

Any idea?

EricClaeys, 2021-09-16:
@laheller, try adding `set ssl:verify-certificate` no to /etc/lftp.conf.
If this fixes your problem, or you already fixed it, please close this issue.  Thanks.

laheller, 2021-09-20:
Hi @EricClaeys 

Problem solved! Can be closed.

BR,

Ladislav