AllskyTeam/allsky #143: Timelapse x264 error

Skippyinspace, 2019-12-12:


Skippyinspace, 2019-12-12:
My Pi 3B+ isnt generating a timelapse. (will it is but its 0 bytes)

From manually running a script I get the following

x264 [error]: malloc of size 25549360 failed
Video encoding failed
[libx264 @ 0x6c41c0] final ratefactor: 31.76
Conversion failed!

thomasjacquin, 2019-12-12:
You have a memory allocation error (malloc). That means the conversion process needs more memory than is physically available on the Pi.
This happens when using a camera with a large sensor. Individual images are large (2000 or 3000px) and the Raspberry Pi can't allocate enough memory.

One solution is to bin your images or resize them before timelapse. The other option is to use a Raspberry Pi 4 which has up to 4GB of RAM.

Skippyinspace, 2019-12-12:
thanks

how do i bin or resize the images and which file do i need to edit.

sorry a bit thick with this.

ian

thomasjacquin, 2019-12-12:
There's a setting in Camera settings called binning. Try using 2 instead of 1.

Skippyinspace, 2019-12-12:
now im a real idiot I cant find camera settings

thomasjacquin, 2019-12-12:
[image]

Skippyinspace, 2019-12-13:
Ok the Gui that i cant get to load, have changed the php etc.

Think im going to do a full reinstall over the weekend.

I take it that the pi noir camera isnt part of the problem

Thanks for your help

Skippyinspace, 2019-12-23:
bought a pi 4

closed 2019-12-23 by Skippyinspace