AllskyTeam/allsky #144: Add support for Pi camera

nikivan1, 2019-12-15:
Pi camera is less sensitive than ASI, but with 3-4 seconds exposures we'd get useful pictures. A lot of people already have it, and even with the added cost of a wide angle lens, it is still much cheaper alternative. Please add support for it. Thanks.

bbillp, 2019-12-16:
I agree because its great to get allsky images while I am imaging withe the ZWO and scope. I can afford just one ZWO so I run raspistill on the allsky Pi 3b+ with 3D printed TV camera lens adapted to the PiCam. ISO 800, 8 seconds, F2.4 is Ok but not fantastic at my location.   Bill acj.

robertpascale, 2020-05-15:
There is a Pull Request #36 for this, with relation to the older cameras.
With the new release of the HQ camera, this may be viable again, especially with the increased resolution, and in particular the exposure time can go up to 200s, 12 megapixels, supports 2x2 binning.

EricClaeys, 2021-09-16:
Closing this issue.  The RPiHQ camera is supported.

closed 2021-09-16 by EricClaeys