AllskyTeam/allsky #148: Bad Flag and Green Square Image

Skippyinspace, 2019-12-23:
Using a Pi 4
A PiNoir Camera - compiled from Pull 36

Getting an output image of a large green Square

Before loading the allsky software camera was working and giving an image

Thanks for the help and merry christmas

Also which file name should I be using
Image.jpg
Liveview-Image.jpg
Resize-Image.jpg

When Running allsky.sh I get the following

OpenCV Error: Bad flag (parameter or structure field) (Unrecognized or unsupported array type) in cvGetMat, file /build/opencv-L65chJ/opencv-3.2.0+dfsg/modules/core/src/array.cpp, line 2501
terminate called after throwing an instance of 'cv::Exception'
  what():  /build/opencv-L65chJ/opencv-3.2.0+dfsg/modules/core/src/array.cpp:2501: error: (-206) Unrecognized or unsupported array type in function cvGetMat

./allsky.sh: line 25:  1972 Aborted                 ./capture $ARGUMENTS
pi@allsky:~/allsky $ 

Allsky.sh file has had the ZWO info taken out about file is below

#!/bin/bash
source /home/pi/allsky/config.sh
source /home/pi/allsky/scripts/filename.sh

echo "Starting allsky camera..."
cd /home/pi/allsky

# Building the arguments to pass to the capture binary
ARGUMENTS=""
KEYS=( $(jq -r 'keys[]' $CAMERA_SETTINGS) )
for KEY in ${KEYS[@]}
do
	ARGUMENTS="$ARGUMENTS -$KEY `jq -r '.'$KEY $CAMERA_SETTINGS` "
done

# When using a desktop environment (Remote Desktop, VNC, HDMI output, etc), a preview of the capture can be displayed in a separate window
# The preview mode does not work if allsky.sh is started as a service or if the debian distribution has no desktop environment.
if [[ $1 == "preview" ]] ; then
	ARGUMENTS="$ARGUMENTS -preview 1"
fi
ARGUMENTS="$ARGUMENTS -daytime $DAYTIME"

echo "$ARGUMENTS">>log.txt

./capture $ARGUMENTS

thomasjacquin, 2019-12-24:
Try this maybe: https://stackoverflow.com/questions/36989044/error-puttext-is-not-a-member-of-cv

Skippyinspace, 2019-12-27:
Thanks that sort of worked but now getting

penCV Error: Bad flag (parameter or structure field) (Unrecognized or unsupported array type) in cvGetMat, file /build/opencv-L65chJ/opencv-3.2.0+dfsg/modules/core/src/array.cpp, line 2501
terminate called after throwing an instance of 'cv::Exception'
  what():  /build/opencv-L65chJ/opencv-3.2.0+dfsg/modules/core/src/array.cpp:2501: error: (-206) Unrecognized or unsupported array type in function cvGetMat

./allsky.sh: line 27:  1577 Aborted                 ./capture $ARGUMENTS

Skippyinspace, 2019-12-31:
looks like a new issue has appeared so closing this one

closed 2019-12-31 by Skippyinspace