AllskyTeam/allsky #149: Turning off Auto Exposure

uzelessknowledge, 2019-12-26:
Using pi2 with latest raspbian (fresh install) and latest allsky (fresh install). When I turn off auto exposure the systems becomes unstable. Images begin to be taken as fast as possible. None are clear or usable. CPU maxes out.

uzelessknowledge, 2019-12-27:
I did a fresh instal on a Pi3 and have the same results with either an ASI224 or ASI120. They both start capturing as fast as they can when exposure is turned off. I turned it off in the GUI and through command line.

thomasjacquin, 2019-12-27:
Can you post your settings here? What's your exposure value?

uzelessknowledge, 2019-12-27:
I tired various exposures. Since it was day time it was using auto automatically.
I did another fresh install on the PI 3. I ran things manually for a bit. Changing the settings.json, stopping and starting all sky each time from command. So far things seem good. I know things are working, but I'd like to re-image one more time and not do the manual testing before installing the GUI.
I'm also going to try again with the PI2 and see.
/Users/Travis/Desktop/Screen Shot 2019-12-27 at 4.11.12 PM.png

thomasjacquin, 2019-12-27:
Yes day time uses auto-exposure. If you want to emulate night time, you can change your longitude temporarily.

Also, if you're NOT using the GUI, settings.json is located at `/home/pi/allsky`.

If you are using the GUI, the program uses settings.json located at `/etc/raspap/`. It gets updated when you save the settings using the GUI. However, you can update it manually as well using a text editor (with sudo).

uzelessknowledge, 2019-12-27:
Yup, I did that. I changed the long in the settings when I was testing manually. After I installed the GUI I just changed it there. 
I’ll update this after I do another image and retest the pi2. 
Thank you!

uzelessknowledge, 2019-12-28:
Well, I can’t get the pi 2 to work. I had it up and got stretch configured, but won’t come back up after a boot. 
The pi 3 is up and I can’t recreate the issues. It’s taking pics fine and not overrunning. 

I’ll keep testing and monitor.

uzelessknowledge, 2019-12-29:
I have tried again with a  Pi 2 and 3. I believe I know what the issue was. The default night delay is '10' which equals 0.01 seconds. The day time delay is 5000 or 5 seconds. I believe the very short night delay was the issue. 
May I suggest that the night delay default be changed to at least 5000 if not 10000.

closed 2019-12-29 by uzelessknowledge

thomasjacquin, 2019-12-29:
The night delay should work fine with a value of 10 milliseconds as long as your exposure is long enough (a few seconds). Otherwise you will cycle really quickly between images and I don't think the RPi will enjoy that.