AllskyTeam/allsky #150: Rpi 4 Buster asi120mm (non S) Image Capture Not working

Hydrazine2501, 2019-12-31:
First off, this is fantastic work, thank you.

My issue is that after upgrading from a 3B+ to a 4B I am no longer getting images from my camera.  The camera worked great under the 3B+ and if I hook it up to a windows host sharpcap works a treat.  So I'm pretty sure the camera is ok.

After installing the software and gui on the 4B the gui comes up just fine, but the image link on the live view page is broken.  Looking in /home/pi/allsky/log.txt I see that one line comes up:

 -angle -6  -autoexposure 1  -autogain 1  -bin 1  -brightness 50  -darkframe 0  -daytimeDelay 5000  -delay 10  -exposure 10000  -filename image.jpg  -flip 0  -fontcolor 255 255 255  -fontline 1  -fontname 0  -fontsize 0.7  -fonttype 0  -gain 50  -gamma 50  -height 0  -latitude 32.5N  -longitude 122.32W  -maxexposure 20000  -maxgain 200  -outlinefont 0  -quality 50  -showDetails 0  -smallfontcolor 0 0 255  -text text  -textx 15  -texty 30  -time 1  -type 0  -usb 40  -wbb 90  -wbr 53  -width 0  -daytime 1

and if I look at /var/log/allsky.log the only thing in it is:

Dec 31 00:57:39 allsky allsky.sh[1816]: can't get debug descriptor: Resource temporarily unavailable

I've read the previous issues on the RPI 4 and Buster.  While others reported the same problems as I am seeing, I'm under the impression (after reading issue #100 and #95) that capture.cpp was updated in order to fix what may be causing this sort of thing, but after several days of trying I can't seem to get this to work.

Does anyone have any thoughts as to what I may be doing wrong, and what I might try?  The logs aren't the most verbose, so it's really hard to see where the application may actually be stuck.

I know that the non-S version of the asi120mm has the odd problem, and honestly I've been due for an upgrade for a long time so I've got an asi290mm on the way.  I'm not sure that would fix this though, and I'd like to figure out what's busted.  Any thoughts or ideas would be appreciated!

Thanks very much!

thomasjacquin, 2019-12-31:
One way to debug the issue is to look for liveview-image.jpg (or whatever image name you chose in the camera settings). The file should be located at `/home/pi/allsky/liveview-image.jpg` and it should update regularly (depends on your exposure settings).
If the file is there then it means it's not a problem with the capture but something to do with the GUI failing to display the image.

Hydrazine2501, 2020-01-01:
It's definitely a problem capturing.  When executing capture by hand it just hangs.  I'll continue to mess with it, and I'll try my asi120mm-S and see if that works.  It may just be that the asi120mm is too dorky.  I know some folks had issues with it.  I'm cautiously optimistic that the asi290mm I've ordered will work ok.  On the raspi 3 I've been using this software in one form or another for years and it worked like a champ.

thomasjacquin, 2020-01-01:
Yes the asi290mm should work fine. 
You might need to use bin 2 or change the resizing ratio because depending on hardware, the timelapse process (ffmpeg) can run out of memory if the individual images are too large.
I think the RPi4 is handling it much better though.

Hydrazine2501, 2020-01-01:
Problem solved-  Camera had to be flashed with the compatible version of the firmware, downloaded from ZWO's site.  Luckily I have a windows PC lying around just for situations like this.

closed 2020-01-01 by Hydrazine2501