AllskyTeam/allsky #152: Temperature readout

WirthmU, 2020-01-26:
Is there an easy way (using python or as a bash script) to readout the sensor temperature of the camera?

Thank you for the support

thomasjacquin, 2020-01-27:
Someone has written python bindings for ZWO cameras. I haven't used it but I assume you can get the sensor temperature by using or modifying this code. https://github.com/stevemarple/python-zwoasi/blob/master/zwoasi/examples/zwoasi_demo.py

WirthmU, 2020-02-26:
I have modified and tried Steve's installation and python script. It worked and I could produce a json file with parameters from the ASI camera. The problem is that if I run the python script it interferes with the allsky programs. Allsky just stops - no pictures taken.
I have looked at the capture.cpp file here. In theory it should be possible to add code to generate a json file and generate it along with the pictures with all the variable readouts from the camera. Unfortunately I am not that good at c++ to do that.
Maybe someone here could do that?

WirthmU, 2020-02-28:
It could be fixed by stopping the allsky service and re-starting it at the end of the pyhton script. It is working now.

[image]

WirthmU, 2020-03-01:
... it is not really fixed! I do get the almost real-time images but all morning functions are haltet (startrails, keogram, timelapses). Most propably intervening with the allsky service also interrupts morning tasks.

I still guess the best way would be to create a json-file with the readout values from the camera in the capture.cpp file.

Would be anyone ready to program that?!

WirthmU, 2020-06-23:
I have finally managed to get a bit used with coding in  C++. I have changed the code of capture.cpp to make a few parameters saved as text whenever a picture is taken. The text file looks like this:

Time: 20200623 15:14:54
ASI_EXPOSURE: 0.000 
ASI_GAIN: 0 
ASI_TEMPERATURE: 85.8

The graphs are then created using rrdtool/cacti as before.

RobertG73, 2020-06-23:
Didn't seen this topic... here is how i export the sensor temp. in Celsius from capture.cpp:
832 //======================== Export sensor temperature ==============================
  			    ofstream myfile;
  			    myfile.open ("sensorC.txt");
  			    myfile << (ltemp / 10);
  			    myfile.close();
837 //=======================================================================

EricClaeys, 2021-09-14:
This issue appears to be resolved.  Please note that in newer versions of capture.cpp, the temperature is written to a file called "temperature.txt".

closed 2021-09-14 by EricClaeys