AllskyTeam/allsky #153: Black live image day time

Skippyinspace, 2020-01-30:
Just set up my asi120mc colour camera, but just getting a black square with info
[image] When changing settings in the gui, I get a good image, then a over exposed and then a black image which it settle on.

Clearly its a setting somewhere but each time I reset to the default and then change my lat long the same happens

thomasjacquin, 2020-01-31:
Is that a fresh install? I'm just thinking there might be 2 processes running at the same time and they are overriding each other.

Skippyinspace, 2020-01-31:
it was but I think ive solved the problem not sure how but it is working now

Thanks for your help

closed 2020-02-13 by Skippyinspace