AllskyTeam/allsky #158: Program not working. code=exited status=127

smetherling, 2020-02-10:
So when I run the program if gives an error.
I've followed every step and didn't see any errors running into the installation.
 Im gonna put a picture where the code is.
[image]

Don't bash on me haha, kinda new into this sort of things.

bern

kpjamro, 2020-02-10:
Line 5 says camera not found.
Can you figure out why?
Ken

smetherling, 2020-02-10:
Well the first attempt I forgot to put the camera in. In the second attempt I plugged it in.

smetherling, 2020-02-10:
Its always give the same error line:
allsky systemd: unit allsky.service entered failed state

smetherling, 2020-02-10:
I use the asi120mc. I've googled if they need any drivers for linux. And they said on the website you don't need any.

kpjamro, 2020-02-10:
My camera is not running this week so i can't look, but what you need to do is one of two things:  
find a *.log file with more info (possibly in /var/log) 
or
run the program from the command line (capture I think it's calle) with the appropriate arguments and see what error messages spit out onto the screen

most likely the problem is some typo in your config file.   Using the web page version of the config will help catch some (but not all) errors.

Ken

thomasjacquin, 2020-02-10:
What Raspberry Pi version are you using? Try a different USB port. Also make sure you’re using the recommended power adapter. If the camera can’t draw enough power, it will not be recognized.

smetherling, 2020-02-10:
> What Raspberry Pi version are you using? Try a different USB port. Also make sure you’re using the recommended power adapter. If the camera can’t draw enough power, it will not be recognized.

SoI looked up my device number. And it had this code a02082, so pi 3 model b. The adapter is the one with the raspberry logo on. My debian version is 8.0.

ckuethe, 2020-02-12:
two things:
* is the camera detected: `# lsusb -v | egrep -i 'zwo|asi'`
* does the `pi` user have permission to open the camera device? you might need to install some udev rules to permit the device properly, eg.
  * https://github.com/openastroproject/openastro/blob/master/udev/70-asi-cameras.rules
  * https://github.com/thomasjacquin/allsky/blob/master/asi.rules

smetherling, 2020-02-19:
Alright so I reinstalled everything. And it works now but it gives the following line when i check the status.
[image]

smetherling, 2020-02-19:
> two things:
>
> * is the camera detected: `# lsusb -v | egrep -i 'zwo|asi'`
> * does the `pi` user have permission to open the camera device? you might need to install some udev rules to permit the [...]

How do you run these commands. I've selected the line for my device. And when i run it, it mentions "attr(idvendor)==03c3,: command not found

ckuethe, 2020-02-19:
Glad that it's working for you. I get that "can't get debug descriptor" too, and my camera works. I think you can safely ignore it.

You should just be able to type `lsusb -v | egrep -i 'zwo|asi'` at the command prompt. When I do that on a machine without the camera connected I see something like this:

```
lsusb -v | egrep -i 'zwo|asi'
can't get debug descriptor: Resource temporarily unavailable
can't get device qualifier: Resource temporarily unavailable
can't get debug descriptor: Resource temporarily unavailable
can't get debug descriptor: Resource temporarily unavailable
can't get device qualifier: Resource temporarily unavailable
can't get debug descriptor: Resource temporarily unavailable
can't get debug descriptor: Resource temporarily unavailable
can't get debug descriptor: Resource temporarily unavailable
can't get device qualifier: Resource temporarily unavailable
can't get debug descriptor: Resource temporarily unavailable
```

And when I do that on a machine with the camera, I get this:

```
lsusb -v | egrep -i 'zwo|asi'
can't get debug descriptor: Resource temporarily unavailable
  iManufacturer           1 ZWO
  iProduct                2 ASI120MM-S
can't get debug descriptor: Resource temporarily unavailable
can't get debug descriptor: Resource temporarily unavailable
can't get debug descriptor: Resource temporarily unavailable
can't get debug descriptor: Resource temporarily unavailable
can't get debug descriptor: Resource temporarily unavailable
can't get device qualifier: Resource temporarily unavailable
can't get debug descriptor: Resource temporarily unavailable
```

So it's pretty clear that I have an ASI120MM-S camera connected.

As for the rules, I'm using custom rules because I'm doing special weird stuff that doesn't play nice with starting allsky from a udev rule. You don't actually run udev rules; udev(7) does that. You just need to make sure that it knows what to do by installing the rules into `/etc/udev/rules.d/` which you seem to have done when you reinstalled. Have a look in there and you'll probably see some asi rules.

smetherling, 2020-02-20:
Alright this afternoon I'll try again. One other thing. When I run the program and try to connect with my laptop to https://allsky.local it mentions I can't connect it. Any idea why?

thomasjacquin, 2020-02-22:
The .local address works if both computers have the _avahi_ service instaled. The RPi should have it running under the name avahi daemon. Apple computers have it running under the name _Bonjour_. On linux you need to install avahi (apt install avahi-daemon) and on Windows I believe you may have it installed if you have iTunes installed or you can just install the Bonjour service independently.

smetherling, 2020-02-25:
Okay! I get the interface now. But i get the next thingies. 
[image]
[image]

As you can see on the live view gives a black screen? And when i go on the dashboard tab it mentions "interface is down". When i run the allsky status thingy it doesnt say any errors.

thomasjacquin, 2020-02-25:
On that first screenshot, the `interface is down` means your wifi is not running. You can see that the wireless info to the right has no details. 
Click on Start wnlan0 to enable the wifi.
Then in the `Configure Wifi` tab, you can choose your network.

smetherling, 2020-02-25:
> On that first screenshot, the `interface is down` means your wifi is not running. You can see that the wireless info to the right has no details.
> Click on Start wnlan0 to enable the wifi.
> Then in the `Configure Wifi` tab, you can choose your network.

If i run by internet cable wont it work?

thomasjacquin, 2020-02-25:
Oh yeah it should work, I thought you were trying to fix the wifi issue.

Regarding the black screen, it looks like the `liveview-image.jpg` doesn't exist. Can you make sure that images are captured properly in `/home/pi/allsky`? The `image.jpg` should refresh periodically.
If it doesn't then we need to know why it's not making new images. You can run `service allsky status` to get more info.

smetherling, 2020-02-25:
Alright. Gonna do that tomorrow. Time for some sleep. The moment it works I'm gonna donate to you to support you. Great help dude!!!

smetherling, 2020-02-26:
Alright so. The interface works. 
[image]

Now when i go to the specific folder you mentioned. I find the jpg file. But it gives a black screen. 

So I've ran the status thingy and this pops up. 
[image]

So close haha to get it working

thomasjacquin, 2020-02-26:
What does `image.jpg` look like? Is it all black? Does it have the time in the upper left corner? Look at the timestamp it was last modified to see if it keeps updating.
Other than that, the admin portal seems to be running fine. The issue seems to be at the capture level.

smetherling, 2020-02-26:
The image had a time on it. Yesterday around 21 oclock.  I will try this evening again when i'm home.

smetherling, 2020-02-26:
> What does `image.jpg` look like? Is it all black? Does it have the time in the upper left corner? Look at the timestamp it was last modified to see if it keeps updating.
> Other than that, the admin portal seems to be running fine. The issue seems to be at the capture level.

[image]
This is the image I get.

smetherling, 2020-02-26:
And if he takes image where does he saves them?

thomasjacquin, 2020-02-26:
That's the default image from the repository. It's not an image created by your camera.
Try running the program manually. First, stop the service (sudo service allsky stop) and then from the allsky directory: `./allsky.sh`
There could be a few reasons why the capture is not starting. One of them is that the time on your RPi could be wrong or thew settings for lat/long for your position are not correct or DAYTIME is not set to 1, etc.

smetherling, 2020-04-01:
Ok. I've checked everything. My location is correct. Daytime is at 1. Time zone is correct on my rpi. I think im gonna give up haha. The live view is still bottle necking me. It gives the broken image thing.

smetherling, 2020-04-01:
[image]
[image]
[image]

These are the settings.

thomasjacquin, 2020-04-01:
I would not even try installing the GUI until the capture is working. Installing the GUI won't fix the issue with the capture.
Make sure images are created in the images folder and then we can move forward to the GUI.
Running `./allsky.sh` is your best option to find out what's happening. Look at the output of that command and paste it here.

EricClaeys, 2021-09-14:
This issue is from an older release of the AllSky software.  If you think it might still be relevant, please test it with the newest version and submit a new issue if needed.   Set the "debuglevel" to 3 to get additional information in /var/log/allsky.log that may help determine what the problem is.

closed 2021-09-14 by EricClaeys