AllskyTeam/allsky #159: Startrails error

across-suffolk, 2020-02-11:
After a particularly clear night there was no skytrails image generated. When I tried to create one manually using this command
`./startrails ./images/20200210/ jpg 0.15 ./images/20200210/startrails.jpg` 
I got the following error:
`...
[956/3548] image-20200210194439.jpg 0.0178334
[957/3548] image-20200210194750.jpg 0.00218969
OpenCV Error: Sizes of input arguments do not match (The operation is neither 'array op array' (where arrays have the same size and type), nor 'array op scalar', nor 'scalar op array') in binary_op, file /build/opencv-L65chJ/opencv-3.2.0+dfsg/modules/core/src/arithm.cpp, line 225                                                    
terminate called after throwing an instance of 'cv::Exception'                                                
  what():  /build/opencv-L65chJ/opencv-3.2.0+dfsg/modules/core/src/arithm.cpp:225: error: (-209) The operation
is neither 'array op array' (where arrays have the same size and type), nor 'array op scalar', nor 'scalar op array' in function binary_op

Aborted`

Is it due to the number of images in the directory or is there something more sinister at play?

thomasjacquin, 2020-02-12:
I would check image 957 or 958 to see if it is corrupted. If so, delete the image and restart the script.

across-suffolk, 2020-02-12:
Thanks Thomas.
It had rained earlier in the evening and I had gone out and cleaned off the dome and tweaked the dew heater - the system was down for around half an hour. Removing all the images before this event allowed the script to run properly. I couldnt see any that were corrupt, just very different from the later clear sky images.  All seems well now.

EricClaeys, 2021-09-14:
This issue appears to have been resolved, so I'm closing it.

closed 2021-09-14 by EricClaeys