AllskyTeam/allsky #161: Adjust image color

dbainbridge, 2020-02-22:
Is there a simple way to adjust the color of the image?  I've noticed the green channel is a bit too strong and need to adjust it.

[image]

thomasjacquin, 2020-02-22:
Nice sky!
The ZWO library doesn't give any option to control the green channel. They only provide access to Red and Blue channels. Have you tried increasing those instead?

dbainbridge, 2020-02-22:
It seems like even when I upped the wbb to 99 and wbr to 63 the red was much more prominent but still had green tinge to it.. 
before:
[image]

after:

[image]

seems like will have to do some post processing?

thomasjacquin, 2020-02-22:
Maybe you could use the -auto-level option when saving the images in saveInageNight.sh.

http://www.imagemagick.org/script/command-line-options.php#auto-level

dbainbridge, 2020-02-22:
Thanks.  I'll explore that.. but also just realized could take advantage of opencv and adjust the color that way before writing.

dbainbridge, 2020-02-22:
I looked closer at your -auto-level suggestion and found a different method called -color-matrix.  It allows you to easily adjust the RGB channels.. i.e. reduce red channel by 20% and green by 30%:
```
convert lv.jpg -color-matrix ' .8 0 0
                                0 .7 0
                                0 0 1 '   lv2.jpg
```
One issue with this is it does affect everything else in the image such as the text but believe there is way to limit it to only the actual image area.  The excessive red in the image is because of light leakage reflecting off of the dome and frost forming.

[image]
[image]

EricClaeys, 2021-09-14:
This issue is from an older release of the AllSky software, and a fix appears to work well enough.  If you think it might still be relevant, please test it with the newest version and submit a new issue if needed.  Thanks

closed 2021-09-14 by EricClaeys