AllskyTeam/allsky #162: Is it possible to install on a VirtualBox?

carballada, 2020-02-26:
I follow all steps, only on the compile part I am lost.
My pi@raspberry:~/allsky $ cat /proc/cpuinfo command return
processor       : 0
vendor_id       : GenuineIntel
cpu family      : 6
model           : 78
model name      : Intel(R) Core(TM) i5-6300U CPU @ 2.40GHz
stepping        : 3
 
what I need to change on .cpp to compile?

thank you!!

thomasjacquin, 2020-02-26:
Sorry, I'm not sure I understand your set up. Are you running this on a Raspbian OS running in VirtualBox?
As you can see here (https://github.com/thomasjacquin/allsky/tree/master/lib) the libraries included are for Armv6 and Armv7 but you can download the x86 one from ZWO's website (https://astronomy-imaging-camera.com/software/ASI_linux_mac_SDK_V1.14.1119.tar.bz2)

The other thing will be to add your platform support in the `Makefile`. Look at ZWO's makefile, they have example to compile on various platforms.

Then you should be able to compile.

carballada, 2020-02-26:
Exactly this, I try to run your software in a virtualized Rapsbian. I will try what you said. thank you

closed 2021-09-14 by EricClaeys