AllskyTeam/allsky #163: Cooler enable for cooled ZWO cameras

peterstellar, 2020-03-01:
Would it be possible to add a function to enable/disable the peltier cooler on ZWO cameras within **allsky**. For long exposures, the benefits of using the cooler are well documented. The addition of this function would be useful for meteor searches and other activities, we are generally using the ASI178MC. Thank you.

thomasjacquin, 2020-03-07:
Hi Peter,

Sorry for the delay. Yes I had a quick look at the documentation that comes the SDK and it seems pretty straightforward to implement. I will rely on owner of cooled cameras for the testing though as I don't own one myself.

peterstellar, 2020-03-11:
Hi Thomas,
The new cooler function appears to work, the fan switches on and if I have the "Show Details" enabled in the Camera Settings, Live View indicates the Sensor temp. in the image. The fan stops for about 4-5 seconds when the config. settings are changed and then restart. As you would expect, the Sensor temp. will rise and then recover after the pause. Operation looks good on an initial test but with a near full moon rising about 9pm, an extended test is quite futile at the moment. Rob is understandably excited and we will keep up the testing. Thank you.
Regards
Peter

thomasjacquin, 2020-03-11:
Hi Peter,
I am glad the new feature is working. The reason for the fan stopping a few seconds is that every time the settings are modified, the service restarts with new arguments. I understand it may not be ideal if the settings are changing a lot but it was designed this way because settings rarely change during the night on allsky cameras.
Please keep me updated once you get a chance to test it under a clear moonless sky.
Thomas

EricClaeys, 2021-09-14:
This issue is from an older release of the AllSky software and the feature was added.  If you think it might still be relevant, please test it with the newest version and submit a new issue if needed.  Thanks

closed 2021-09-14 by EricClaeys