AllskyTeam/allsky #165: Black/No live view on GUI

1Zach1, 2020-03-14:
Hi,

Thanks for all you do with the AllSky. I am attempting to set mine up but running into an issue I can’t seem to figure out. I have checked the previous issues and don’t seem to have come up with any solutions for what is happening. 

When I run the GUI install everything seems to go fine, but I get either a black image (see picture) or a broken image icon. There also don’t seem to be any pictures being saved, black or otherwise. 

Running the install from the command line without the GUI and everything seems fine, I can view a live preview and the image comes up normal, so I know the camera is connecting and running. 

I have restarted everything, reformatted the card and reinstalled everything (a few times while trying to troubleshoot). I am using all the default settings, and have attempted to manually run using the GUI with the same result. 

Any suggestions?

ASI120MC-S
Raspberry Pi 3 B+

[image]

Thanks 
Zach

thomasjacquin, 2020-03-15:
If the images are not saved, the GUI will not display anything.
There could be a few reasons why the images don't save. Did you adjust the lat/long for your location? The images only save at night. Night and day are calculated based on your GPS coordinates and the time on your RPi.
Maybe post your settings and I can take a look.

1Zach1, 2020-03-15:
I have been using the default settings up to this point, but changed the location in the GUI and on my RPi to a random night location (see settings picture attached). Still getting the black/broken image and no saved pictures. 
[image]

thomasjacquin, 2020-03-15:
The image on the GUI will stay black until the capture problem is resolved. Can you post the output of `./allsky.sh`? Remember to stop the service first (sudo service allsky stop).

1Zach1, 2020-03-15:
When I run sudo service allsky stop I get “ Failed to stop allsky.service: Unit allsky.service not loaded

I ran ./allsky.sh after stopping in the GUI and got “bash: ./allsky.sh: No such file or directory”

thomasjacquin, 2020-03-16:
Well that's not normal. There should be a file called `allsky.sh` in the /home/pi/allsky directory.
Are you running the command from the allsky directory?

closed 2020-03-22 by 1Zach1