AllskyTeam/allsky #167: Error in the picture with ASI120MC

DM2NT, 2020-03-17:
Hi there,

I use a Raspberry Pi 4 with 1Gb Ram and an ASI120MC.
There are strange errors in some pictures.
I attach a picture to show what I mean.
In addition, the recording does not work during the day, although set to "true" in the config. No pictures are saved during the day. Alledings that worked 3 days ago.
I also flashed the alternative firmware onto the ASI120MC. It didn't work without it.

Tino

[https://www.directupload.net/file/d/5761/glfjcqip_jpg.htm](url)
[https://www.directupload.net/file/d/5761/g2d6484r_jpg.htm](url)

thomasjacquin, 2020-03-19:
Someone fixed this issue for ASI120MC. I haven't merged the code because it was causing a delay in capture for other cameras but you can find the changes here:
https://github.com/thomasjacquin/allsky/pull/154/files#diff-4b2866459accc9735665bfd8158dafbf

You can get the file here:
https://raw.githubusercontent.com/liangchen-harold/allsky/master/capture.cpp

Replace your capture.cpp and re-compile by running with `make capture`

marcfrincu, 2020-03-19:
I also installed today the scripts (the new capture.cpp file indicated here) but no image is being saved.  I have a Pi4 B with 2GB RAM and an ASI120MC. ./allsky.sh indicates that the camera is OK but the program freezes at "Press Ctrl+C to stop". It does not even respond to Ctrl+C. Seems it is stuck calling some functions but I cannot tell which. I tried placing some printfs in the capture.cpp and recompile it but nothing shows up.  I am trying to save Raw8 PNG images. Tried with jpg and other formats as well without success. Any clues?

thomasjacquin, 2020-03-21:
@DM2NT By the way, daytime=1 means that it streams during the day but it doesn't capture. Capture only happens at night.

thomasjacquin, 2020-03-21:
@marcfrincu Is there anything logged when you run `service allsky status`?
Can you post your settings and a screenshot of the output of `./allsky.sh`?

marcfrincu, 2020-03-25:
Hello,

I have finally managed to get some output by updating the ASI12MC FW by using the compatible devices firmware version. I have updated it from Windows. However now the program (starting from command line) behaves quite strange. When saving jpeg images with brightness=0 they are very bright (not expected) but with the same settings the png images are almost black (which is to be expected). I will continue playing around with the settings.

The first three lines of the output are:

./allsky.sh 
can't get debug descriptor: Resource temporarily unavailable
Starting allsky camera...
Saving file to image
...
and then info about the camera, the settings file and the periodic capture.

phil-zenger, 2020-05-15:
I am still getting banding issues (vertical lines) on some of my pics. Here is my mp4 from last night:

https://youtu.be/-q8sDmtODxA

I am using:
Raspberry Pi 3 Model B Rev 1.2
ASI120MC-S
USB 3.0 cable
I have changed my USB settings from 40 to 90, but get same problem.  I have been in contact with ZWO support to try to get them to help me see debug logs. They sent me the following SDK Debug tool for Windows (attached program and Manual). Does anybody know how to turn on Debug on the Pi?
[Dbgview.zip](https://github.com/thomasjacquin/allsky/files/4635176/Dbgview.zip)
[ZWO Catch  SDK LOG manul.zip](https://github.com/thomasjacquin/allsky/files/4635177/ZWO.Catch.SDK.LOG.manul.zip)

Dentonknifeworks, 2020-05-17:
> I am still getting banding issues (vertical lines) on some of my pics. Here is my mp4 from last night:
>
> https://youtu.be/-q8sDmtODxA
>
> I am using:
> Raspberry Pi 3 Model B Rev 1.2
> ASI120MC-S
> USB 3.0 [...]
I have seen the same issue and I have the same setup as you Only my vertical lines are on the right side. I have not figured it out yet. For me it has been random. I have set it back to default settings and I have not seen the issue sense then. I am going to let it run for a week before I tweet anything then do one thing at a time.
 I had a quick look at your Dbgview.zip and thats just not going to work on a PI.

phil-zenger, 2020-05-17:
I have my settings.json option FLIP = 1 so that my images are oriented with East on the right side. So actually are vertical lines occur on the same side.

deltaf508, 2020-07-20:
> Someone fixed this issue for ASI120MC. I haven't merged the code because it was causing a delay in capture for other cameras but you can find the changes here: [...]

Pi Newbie here and I'm seeing this split screen issue.  I have an ASI120MC as well.  is replacing the capture.cpp file with the one provided here still the best step.  If so can someone tell me how to "recompile with make capture" in more clear terms.  

Thanks!

Torakin9366, 2020-08-02:
Ich habe ebenfals diese Fehler im Bild, aber nur am Tage und im Livebild

Torakin9366, 2020-08-02:
[image]
[image]
[image]
[image]
Was kann ma dagegen machen, Asi 120 MC-S passiert aber nur am Tage. in der Nacht wenn die Aufnahme Startet gibt es keine Fehler, es muss was mit der Aufnahme zu tun haben

Torakin9366, 2020-08-02:
[image]
 
so sieht es aus wenn die Aufnahme läuft, ich denke mal das es was mit den Expusure zu tun hat unter 1 sek Fehler im Bild, über 1 sek keine Fehler

[image]

EricClaeys, 2021-09-14:
This issue is from an older release of the AllSky software.  If you think it might still be relevant, please test it with the newest version and submit a new issue if needed.  Thanks

closed 2021-09-14 by EricClaeys