AllskyTeam/allsky #168: Question - Other Sensors?

kenbiba, 2020-04-02:
I am bringing up a new AllSky installation - and I see in some of the existing installations other environmental sensors (magnetic, humidity, temperature, etc) are displayed on the web page.   Without looking at the code - did not see provision for those sensors in the documentation.

Are these custom installations?

Love to add weather and mag sensors to my installation.

thomasjacquin, 2020-04-03:
Yes those are custom additions by members. This is either local information from temperature sensors or data gathered from online services such as weather underground, NOAA, etc.

EricClaeys, 2021-09-14:
Many people use the "Text Overlay" and/or "Extra Text File" settings on the GUI to add the additional information (the "text" and "extratext" settings if not using the GUI).

closed 2021-09-14 by EricClaeys