AllskyTeam/allsky #169: Creating timelapse does not work

Duke4ever73, 2020-04-13:
Hi,
my first installation a few weeks ago created error-free timelapse files. I did a new installation with the same hardware (Raspi 4, 4GB, zwo asi 290mc)... and now the created timelapse files are 0MB in size. Manual creation of the timelapse generated an error message at the end of the program.

_libpostproc 55. 3.100 / 55. 3.100
Input #0, image2, from 'images/20200408/sequence/%04d.jpg':
Duration: 00:01:09.68, start: 0.000000, bitrate: N/A
Stream #0:0: Video: mjpeg, yuvj420p(pc, bt470bg/unknown/unknown), 1936x1096 [SAR 1:1 DAR 242:137], 25 fps, 25 tbr, 25 tbn, 25 tbc
Stream mapping:
Stream #0:0 -> #0:0 (mjpeg (native) -> h264 (h264_omx))
Press [q] to stop, [?] for help
[swscaler @ 0xcdc510] deprecated pixel format used, make sure you did set range correctly
[h264_omx @ 0xc2ab40] Using OMX.broadcom.video_encode
[h264_omx @ 0xc2ab40] OMX error 80001000
[h264_omx @ 0xc2ab40] err 80001000 (-2147479552) on line 561
Error initializing output stream 0:0 -- Error while opening encoder for output stream #0:0 - maybe incorrect parameters such as bit_rate, rate, width or height
Conversion failed!
* Deleting sequence
* Timelapse was created_

Images are ok, keogramm works, another new installation >> same Problems.

any ideas?

thank you and best regards

thomasjacquin, 2020-04-14:
Hi,

I added an FPS parameter in config.sh that is used in timelapse.sh.
If you copied your config files over from the old installation, you're probably missing this parameter.

cbassa, 2020-04-14:
[This thread](https://www.raspberrypi.org/forums/viewtopic.php?t=188317) indicates that this error is related to the GPU/CPU memory split, and that the standard 64MB GPU memory may not be enough. 

I had the same issue on my system (RPi3B+, ASI178MC), and increasing the GPU memory to 128MB did remove the error message, though `ffmpeg` was stuck, probably because the individual JPG images were too large (3096x2080). After resizing the images by 50% the `timelapse.sh` script worked as expected.

Duke4ever73, 2020-04-14:
Hi, guys,
Thanks for your help but I could not solve the problem yet.
My zwo asi290 has only a resolution of 1936×1096 and the file size is only about 450kb. I do not think it is a memory problem. Nevertheless I have set the gpu value to 256mb. >> without success.
As I said before, in February it still worked fine with the same hardware. What I am trying to do now is to track all changes since the beginning of March. I have already noticed that new Asi-SDK have been implemented. Could this be a possible cause?? I have now saved the old SDK files into the directories.
Let's see what happens... I will report.
(Sorry for my bad English, I am an old man from Germany)

Translated with www.DeepL.com/Translator (free version)

thomasjacquin, 2020-04-14:
@Duke4ever73 I changed the codec used for the timelapse generation. Try the old one and see if it works:
https://github.com/thomasjacquin/allsky/commit/a354ac6995dc9a83270f8377375d68b2fc2459dd#diff-fc79c152c3353a80f6fa1468c3c903ea

Duke4ever73, 2020-04-14:
I changed the codec in timelapse.sh !!
Now it works!! Very good. Thank you very much.

A cup of coffee is yours :-)

joe-fenton, 2020-04-21:
@thomasjacquin changing back to the old codec fixed the same problem for me too. Also on the Raspberry Pi 4, with ZWO 290MC camera. Is it just a missing codec issue or something else?

thomasjacquin, 2020-04-22:
@joe-fenton I am trying to figure that one out. Looks like `omx` was enabled automatically in my version of `ffmpeg` but it seems that it's not the case for everyone.

Can you run this command in a terminal and copy the output?
```
ffmpeg -encoders | grep omx
```

Mine returns this
```
ffmpeg version 4.1.4-1+rpt1~deb10u1 Copyright (c) 2000-2019 the FFmpeg developers
  built with gcc 8 (Raspbian 8.3.0-6+rpi1)
  configuration: --prefix=/usr --extra-version='1+rpt1~deb10u1' --toolchain=hardened --libdir=/usr/lib/arm-linux-gnueabihf --incdir=/usr/include/arm-linux-gnueabihf --arch=arm --enable-gpl --disable-stripping --enable-avresample --disable-filter=resample --enable-avisynth --enable-gnutls --enable-ladspa --enable-libaom --enable-libass --enable-libbluray --enable-libbs2b --enable-libcaca --enable-libcdio --enable-libcodec2 --enable-libflite --enable-libfontconfig --enable-libfreetype --enable-libfribidi --enable-libgme --enable-libgsm --enable-libjack --enable-libmp3lame --enable-libmysofa --enable-libopenjpeg --enable-libopenmpt --enable-libopus --enable-libpulse --enable-librsvg --enable-librubberband --enable-libshine --enable-libsnappy --enable-libsoxr --enable-libspeex --enable-libssh --enable-libtheora --enable-libtwolame --enable-libvidstab --enable-libvorbis --enable-libvpx --enable-libwavpack --enable-libwebp --enable-libx265 --enable-libxml2 --enable-libxvid --enable-libzmq --enable-libzvbi --enable-lv2 --enable-omx --enable-openal --enable-opengl --enable-sdl2 --enable-omx-rpi --enable-mmal --enable-libdc1394 --enable-libdrm --enable-libiec61883 --enable-chromaprint --enable-frei0r --enable-libx264 --enable-shared
  libavutil      56. 22.100 / 56. 22.100
  libavcodec     58. 35.100 / 58. 35.100
  libavformat    58. 20.100 / 58. 20.100
  libavdevice    58.  5.100 / 58.  5.100
  libavfilter     7. 40.101 /  7. 40.101
  libavresample   4.  0.  0 /  4.  0.  0
  libswscale      5.  3.100 /  5.  3.100
  libswresample   3.  3.100 /  3.  3.100
  libpostproc    55.  3.100 / 55.  3.100
 V..... h264_omx             OpenMAX IL H.264 video encoder (codec h264)
```

joe-fenton, 2020-04-22:
Sure, mine returned a lot more stuff before that information...
`ffmpeg version 4.1.4-1+rpt7~deb10u1 Copyright (c) 2000-2019 the FFmpeg developers
  built with gcc 8 (Raspbian 8.3.0-6+rpi1)
  configuration: --prefix=/usr --extra-version='1+rpt7~deb10u1' --toolchain=hardened --incdir=/usr/include/arm-linux-gnueabihf --enable-gpl --disable-stripping --enable-avresample --disable-filter=resample --enable-avisynth --enable-gnutls --enable-ladspa --enable-libaom --enable-libass --enable-libbluray --enable-libbs2b --enable-libcaca --enable-libcdio --enable-libcodec2 --enable-libflite --enable-libfontconfig --enable-libfreetype --enable-libfribidi --enable-libgme --enable-libgsm --enable-libjack --enable-libmp3lame --enable-libmysofa --enable-libopenjpeg --enable-libopenmpt --enable-libopus --enable-libpulse --enable-librsvg --enable-librubberband --enable-libshine --enable-libsnappy --enable-libsoxr --enable-libspeex --enable-libssh --enable-libtheora --enable-libtwolame --enable-libvidstab --enable-libvorbis --enable-libvpx --enable-libwavpack --enable-libwebp --enable-libx265 --enable-libxml2 --enable-libxvid --enable-libzmq --enable-libzvbi --enable-lv2 --enable-omx --enable-openal --enable-opengl --enable-sdl2 --enable-omx-rpi --enable-mmal --enable-neon --enable-rpi --enable-libdc1394 --enable-libdrm --enable-libiec61883 --enable-chromaprint --enable-frei0r --enable-libx264 --enable-shared --libdir=/usr/lib/arm-linux-gnueabihf --cpu=arm1176jzf-s --arch=arm
  WARNING: library configuration mismatch
  avutil      configuration: --prefix=/usr --extra-version='1+rpt7~deb10u1' --toolchain=hardened --incdir=/usr/include/arm-linux-gnueabihf --enable-gpl --disable-stripping --enable-avresample --disable-filter=resample --enable-avisynth --enable-gnutls --enable-ladspa --enable-libaom --enable-libass --enable-libbluray --enable-libbs2b --enable-libcaca --enable-libcdio --enable-libcodec2 --enable-libflite --enable-libfontconfig --enable-libfreetype --enable-libfribidi --enable-libgme --enable-libgsm --enable-libjack --enable-libmp3lame --enable-libmysofa --enable-libopenjpeg --enable-libopenmpt --enable-libopus --enable-libpulse --enable-librsvg --enable-librubberband --enable-libshine --enable-libsnappy --enable-libsoxr --enable-libspeex --enable-libssh --enable-libtheora --enable-libtwolame --enable-libvidstab --enable-libvorbis --enable-libvpx --enable-libwavpack --enable-libwebp --enable-libx265 --enable-libxml2 --enable-libxvid --enable-libzmq --enable-libzvbi --enable-lv2 --enable-omx --enable-openal --enable-opengl --enable-sdl2 --enable-omx-rpi --enable-mmal --enable-neon --enable-rpi --enable-libdc1394 --enable-libdrm --enable-libiec61883 --enable-chromaprint --enable-frei0r --enable-libx264 --libdir=/usr/lib/arm-linux-gnueabihf/neon/vfp --cpu=cortex-a7 --arch=armv6t2 --disable-thumb --enable-shared --disable-doc --disable-programs
  avcodec     configuration: --prefix=/usr --extra-version='1+rpt7~deb10u1' --toolchain=hardened --incdir=/usr/include/arm-linux-gnueabihf --enable-gpl --disable-stripping --enable-avresample --disable-filter=resample --enable-avisynth --enable-gnutls --enable-ladspa --enable-libaom --enable-libass --enable-libbluray --enable-libbs2b --enable-libcaca --enable-libcdio --enable-libcodec2 --enable-libflite --enable-libfontconfig --enable-libfreetype --enable-libfribidi --enable-libgme --enable-libgsm --enable-libjack --enable-libmp3lame --enable-libmysofa --enable-libopenjpeg --enable-libopenmpt --enable-libopus --enable-libpulse --enable-librsvg --enable-librubberband --enable-libshine --enable-libsnappy --enable-libsoxr --enable-libspeex --enable-libssh --enable-libtheora --enable-libtwolame --enable-libvidstab --enable-libvorbis --enable-libvpx --enable-libwavpack --enable-libwebp --enable-libx265 --enable-libxml2 --enable-libxvid --enable-libzmq --enable-libzvbi --enable-lv2 --enable-omx --enable-openal --enable-opengl --enable-sdl2 --enable-omx-rpi --enable-mmal --enable-neon --enable-rpi --enable-libdc1394 --enable-libdrm --enable-libiec61883 --enable-chromaprint --enable-frei0r --enable-libx264 --libdir=/usr/lib/arm-linux-gnueabihf/neon/vfp --cpu=cortex-a7 --arch=armv6t2 --disable-thumb --enable-shared --disable-doc --disable-programs
  avformat    configuration: --prefix=/usr --extra-version='1+rpt7~deb10u1' --toolchain=hardened --incdir=/usr/include/arm-linux-gnueabihf --enable-gpl --disable-stripping --enable-avresample --disable-filter=resample --enable-avisynth --enable-gnutls --enable-ladspa --enable-libaom --enable-libass --enable-libbluray --enable-libbs2b --enable-libcaca --enable-libcdio --enable-libcodec2 --enable-libflite --enable-libfontconfig --enable-libfreetype --enable-libfribidi --enable-libgme --enable-libgsm --enable-libjack --enable-libmp3lame --enable-libmysofa --enable-libopenjpeg --enable-libopenmpt --enable-libopus --enable-libpulse --enable-librsvg --enable-librubberband --enable-libshine --enable-libsnappy --enable-libsoxr --enable-libspeex --enable-libssh --enable-libtheora --enable-libtwolame --enable-libvidstab --enable-libvorbis --enable-libvpx --enable-libwavpack --enable-libwebp --enable-libx265 --enable-libxml2 --enable-libxvid --enable-libzmq --enable-libzvbi --enable-lv2 --enable-omx --enable-openal --enable-opengl --enable-sdl2 --enable-omx-rpi --enable-mmal --enable-neon --enable-rpi --enable-libdc1394 --enable-libdrm --enable-libiec61883 --enable-chromaprint --enable-frei0r --enable-libx264 --libdir=/usr/lib/arm-linux-gnueabihf/ [... cut ...]
  libavfilter     7. 40.101 /  7. 40.101
  libavresample   4.  0.  0 /  4.  0.  0
  libswscale      5.  3.100 /  5.  3.100
  libswresample   3.  3.100 /  3.  3.100
  libpostproc    55.  3.100 / 55.  3.100
 V..... h264_omx             OpenMAX IL H.264 video encoder (codec h264)
`
the 'omx' part on the last line was oddly coloured red? I'm not sure why there was so many libraries before the info at the end. I haven't installed too much other stuff on my Pi 4. I am in the UI though, rather than booting just to terminal.

joe-fenton, 2020-04-22:
I have just spotted the WARNING library configuration mismatch in the middle of all that. I've tried googling solutions, and the one I tried was to remove
`sudo apt remove libavcodec-ffmpeg-extra56`
but that didn't work for me, I guess that tip was for Ubuntu and this is Rasbian. I'm wondering if a library needs to be recompiled.

rogergau, 2020-04-26:
I have the same problem. I upgraded just a few days ago and haven't had a timelapse since. Pi4. 

I've run the command:

`ffmpeg -encoders | grep omx`

which returned:

```
ffmpeg version 3.2.14-1~deb9u1+rpt1 Copyright (c) 2000-2019 the FFmpeg developers
  built with gcc 6.3.0 (Raspbian 6.3.0-18+rpi1+deb9u1) 20170516
  configuration: --prefix=/usr --extra-version='1~deb9u1+rpt1' --toolchain=hardened --libdir=/usr/lib/arm-linux-gnueabihf --incdir=/usr/include/arm-linux-gnueabihf --enable-gpl --disable-stripping --enable-avresample --enable-avisynth --enable-gnutls --enable-ladspa --enable-libass --enable-libbluray --enable-libbs2b --enable-libcaca --enable-libcdio --enable-libebur128 --enable-libflite --enable-libfontconfig --enable-libfreetype --enable-libfribidi --enable-libgme --enable-libgsm --enable-libmp3lame --enable-libopenjpeg --enable-libopenmpt --enable-libopus --enable-libpulse --enable-librubberband --enable-libshine --enable-libsnappy --enable-libsoxr --enable-libspeex --enable-libssh --enable-libtheora --enable-libtwolame --enable-libvorbis --enable-libvpx --enable-libwavpack --enable-libwebp --enable-libx265 --enable-libxvid --enable-libzmq --enable-libzvbi --enable-omx --enable-omx-rpi --enable-mmal --enable-openal --enable-opengl --enable-sdl2 --enable-libdc1394 --enable-libiec61883 --arch=armhf --enable-chromaprint --enable-frei0r --enable-libopencv --enable-libx264 --enable-shared
  libavutil      55. 34.101 / 55. 34.101
  libavcodec     57. 64.101 / 57. 64.101
  libavformat    57. 56.101 / 57. 56.101
  libavdevice    57.  1.100 / 57.  1.100
  libavfilter     6. 65.100 /  6. 65.100
  libavresample   3.  1.  0 /  3.  1.  0
  libswscale      4.  2.100 /  4.  2.100
  libswresample   2.  3.100 /  2.  3.100
  libpostproc    54.  1.100 / 54.  1.100
 V..... h264_omx             OpenMAX IL H.264 video encoder (codec h264)
```

I have edited timelapse.sh to set the codec back and will see if it works overnight toinght.

Roger

thomasjacquin, 2020-04-26:
Thanks Roger, I have reverted the coded on the repository. Let me know if switching back to the old code worked for you

dts350z, 2020-07-17:
I have a new install as of 07-15-2020. Pi 3B+ zwo ASI178mm and can't get timelapses to work. I can't tell from the above thread what to try, since I have the latest code, but I did try different gpu mem sizes.

No mater what timelapse either fails with malloc error or Pi locks up. I can see the Pi going into swap.

How are people resizing the images before creating timelapse?

dts350z, 2020-07-17:
So the memory management for the mpeg2 encoder is much better. It does go up and up, but stops increasing as soon as any swap is detected, so proceeds to the end.

ffmpeg -y -f image2 -r 25 -i images/$1/sequence/%04d.$EXTENSION -vcodec mpeg2video  -b 2500k -bt 300k -pix_fmt yuv420p images/$1/allsky-$1.mp4

The web form won't play it, so maybe the .mp4 extension is wrong, but if I download it I can play it in VLC. Still experimenting...

EricClaeys, 2021-09-14:
This issue is from an older release of the AllSky software.  If you think it might still be relevant, please test it with the newest version and submit a new issue if needed.  Thanks

closed 2021-09-14 by EricClaeys