AllskyTeam/allsky #170: libASICamera2.a versions

cbassa, 2020-04-14:
The software ships with two versions of `libASICamera2.a`, an `armv6` and and `armv7` one. Though the recent RPi's have the `armv7` architecture, it appears the `armv6` version of `libASICamera2.a` provides a color balance which is more in line with that used by the `ASIStudio` software from ZWO on `x86` machines.

Here are images taken with an ASI178MC using `armv6`, `armv7` and `x86_64` versions of `libASICamera2.a`, all with red balance 70, blue balance 90.
[image]
[image]
[image]

Is this issue seen with other ASI color cameras as well, or does it only exist with the ASI178MC? You can test this by recompiling `capture.cpp` and linking it against either `lib/armv6/libASICamera2.a` or `lib/armv7/libASICamera2.a`.

lumdiniz, 2020-04-25:
como vincular? qual o comando? tenho uma asi224mc

EricClaeys, 2021-09-14:
This issue is from an older release of the AllSky software.  If you think it might still be relevant, please test it with the newest version and submit a new issue if needed.  Thanks

closed 2021-09-14 by EricClaeys