AllskyTeam/allsky #171: Constellations Overlay not working anymore

hdiessner, 2020-04-15:
When using the web gui of the allsky cam, you can switch on and off "constellations overlay". In previous times, the stars and constellations were displayed whe turned on - now only a coordinates grid is displayed, no more stars.
I checked it not only on my own allsky cam but also on others (like Thomas, yours), same here.
Any obvious reason or a hint where to search for the potential break?

thomasjacquin, 2020-04-16:
Hi Herwig,
Is your browser in a different language than English? Like German maybe? I believe VirtualSky is the issue here. It doesn't really support other languages or maybe I need to bring in some lang packages from their project.
Here's how it looks like on my end.

[image]

hdiessner, 2020-04-16:
Hi Thomas,
you are totally right! I would never have thought of that! It was the language setting of the browser, indeed! Thank you!

closed 2020-04-16 by hdiessner