AllskyTeam/allsky #172: Problems playing video from images page

pterodyne, 2020-04-16:
When I go to the images page and click on the video icon for a particular day I see this in firefox:
No video with supported format and MIME type found

I tried in vivaldi browser (chromium based) and the video box appears but is black.  Clicking on play does nothing.

System: Pi4/4gb
Latest build on debian 10
ASI1600M camera

thomasjacquin, 2020-04-16:
Could you take a look at this change I made and replace the latest code by the old one?
https://github.com/thomasjacquin/allsky/commit/a354ac6995dc9a83270f8377375d68b2fc2459dd#diff-fc79c152c3353a80f6fa1468c3c903ea

I have replaced the codec for a new one. That might be the issue.
Then you can try regenerating the timelapse for that night.
`./timelapse.sh 20200415`

pterodyne, 2020-04-17:
I edited the script to match the old variable.  Here is what I get (still not working)

```
Output #0, mp4, to 'images/20200415/allsky-20200415.mp4':
  Metadata:
    encoder         : Lavf58.20.100
    Stream #0:0: Video: h264 (libx264) (avc1 / 0x31637661), yuv420p, 4656x3520 [SAR 1:1 DAR 291:220], q=-1--1, 2000 kb/s, 25 fps, 12800 tbn, 25 tbc
    Metadata:
      encoder         : Lavc58.35.100 libx264
    Side data:
      cpb: bitrate max/min/avg: 0/0/2000000 buffer size: 0 vbv_delay: -1
x264 [error]: malloc of size 50974512 failedme=00:00:00.00 bitrate=N/A speed=   0x
Video encoding failed
[libx264 @ 0x9d9380] final ratefactor: 34.82
Conversion failed!
* Deleting sequence
* Timelapse was created

```

thomasjacquin, 2020-04-17:
Oh sorry I didn't notice you have an ASI1600. This camera has a huge sensor so your problem here is that `ffmpeg` is hitting a memory cap because it can't allocate enough memory to build the timelapse.
There are 2 options for you:
- Use bin 2 or more to reduce image size
- Reduce image size before creating the timelapse.

For the second option, you can add this line in ./timelapse.sh before the `gawk` command. Use the percentage you desire. Keep in mind that this will replace the images. If you want to keep the originals, you'll need to make a copy of the images first.
```
mogrify -resize 30% images/201808/17/*.jpg
```

pterodyne, 2020-04-22:
There isn't a way with a 4GB Pi 4 to create a full resolution video with the asi 1600?  It seems like it should be able to handle it with 4GB of RAM.  Im running it with the mogrify command at the moment to see.  It's taking forever and the cpu is getting very hot

kpjamro, 2020-04-22:
I run one of the ASI cameras that has 6 megapixel resolution (12 bits) monochrome.  That seems to put the RPI4 at its limit.  It will work with 2 GB but just barely, so I use 4 GB and it takes until mid day to make the time lapse.  Your camera is 16 megapixel and that is more than double what I have.  I think the only way you will get it to work is with 2x2 binning.

good luck

dbainbridge, 2020-04-22:
The other option that I have thought of since my RPi3 can't handle generating full size time lapse video even for my ASI178 is to upload the full resolution images to a server and have the server itself create the time lapse.

kpjamro, 2020-04-22:
Done that. It works. A big hulking desktop might be some 20 times as fast as a pi and have plenty of RAM.

AstroSmoke, 2020-04-25:
Hi, I've just built a fresh build of the allsky and also had to revert the codec as discussed in this thread to make the time lapse video work. I did then notice the star trails failed, but need to allow another nights operation to confirm this behavior.

Pi4B 4gb, 64mb SDcard
Latest firmware and Raspbian  (3 days old)
Fresh install of allsky + reverted codec

thomasjacquin, 2020-04-25:
Ok thanks for the report, I will revert if and investigate what dependencies to bring to use the new codec.

EricClaeys, 2021-09-14:
This issue is from an older release of the AllSky software.  If you think it might still be relevant, please test it with the newest version and submit a new issue if needed.  Thanks

closed 2021-09-14 by EricClaeys