AllskyTeam/allsky #173: Wired connection instead of wireless

maserowik, 2020-04-20:
Greetings. I am using a POE for my raspberry pi.  Is there a way to have AllSky use the Ethernet instead of the Wireless connection?

Thanks
Mike

dwormuth, 2020-04-20:
Works by default as long as you enable wired ethernet in the operating
system. That's how my system operates.

maserowik, 2020-04-20:
[image]

not following this.  I have a valid ip address and 192.168.1.25 but it not showing in the screen shot below in the config

dwormuth, 2020-04-20:
Troubleshoot tips:
https://www.instructables.com/id/Configuring-the-Raspberry-Pi-ethernet-port-for-rem/

https://www.mathworks.com/help/supportpkg/raspberrypi/ug/getting-the-raspberry_pi-ip-address.html

https://raspberrypi.stackexchange.com/questions/37920/how-do-i-set-up-networking-wifi-static-ip-address

These sites should get you up and running. It's a Raspberry Pi issue, not
Allsky

AstroSmoke, 2020-04-25:
Sorry I am not convinced, I have the same behavior where the ip does not show in the interface and says that is down. Yet the ifconfig command on the pi displays ip information.

Maybe allsky is calling the ifconfig too early prior to the ip address being assigned by the network.

Using a Pi4B 4GB, fresh load of firmware, raspbian and allsky three days ago.
I'm just testing on Ethernet in my office with camera pointing the window.

thomasjacquin, 2020-04-25:
allsky is just listing the wlan0 interface which is the wifi interface. If you're using ethernet, it is normal that it reports that wlan0 is down.

maserowik, 2020-04-25:
Any way to add the ethernet also?

 

This email has been checked for viruses by Avast antivirus software.
https://www.avast.com/antivirus

pterodyne, 2020-05-26:
+1 on this request.  I use POE and Ethernet in my allsky.  Doesn't really matter but it would be nice if it used whatever interface was live.

closed 2020-08-06 by maserowik

pterodyne, 2020-08-17:
This still seems like a valid request.  Why only list the wifi?  Many of us use ethernet.  IMHO it should also list a valid ethernet status if it is being used.  Also it claims to use "information provided by ifconfig and iwconfig".  If I do an ifconfig, I can see my eth0 interface details.  Just my .02$.  Now Ill get back in my lane as a consumer of free software that I lack the skill to contribute to. (Other than asking for stuff). :)